home-assistant/operating-system #461: RPi 4 Wifi Connection Problems

fjfricke, 2019-08-28:
I tried the new release 3.4 with the RPi 4 (32 Bit). The Raspberry connects to the wifi on the very first boot but not afterwards. If I disconnect the device from the power supply and connect it again, it cannot connect again.

fjfricke, 2019-08-28:
It could also be the case that the device doesn't boot at all. Will check if it works with LAN tomorrow.

pvizeli, 2019-08-29:
Check the content of `/mnt/overlay/etc/NetworkManager`

fjfricke, 2019-08-29:
There is one file in system-connections/my-network which is the same as the one I created for the fist boot.
It's wiered: This time after I reformatted my sd card, it connected to the wifi twice but not after that.

pvizeli, 2019-08-29:
what is the log of NetworkManager?

fjfricke, 2019-08-29:
Where do I find the log file? I couldn't find it on my device..

pvizeli, 2019-08-29:
`journalctl`

fjfricke, 2019-08-29:
Apparently the problem was that I put an hassio backup into the backup folder on the SD card and it was not booting afterwards. I did it via Samba and now it is working.

closed 2019-08-30 by pvizeli