home-assistant/operating-system #462: There is a Home Assistant / Hass.io / HassOS. Which kind of the update do you means?

iannecj, 2019-08-29:
There is a Home Assistant / Hass.io / HassOS. Which kind of the update do you means?

Edit: Don't post on PRs. You can open a issues and provide more details.

_Originally posted by @pvizeli in https://github.com/home-assistant/hassos/pull/459#issuecomment-526114394_

iannecj, 2019-08-29:
Apologies for posting in wrong place, your note helped though...I had updated HA and not hassio (I am a newbie HA user and a newbie git user).

I can confirm 4GB now showing on my sysmon sensor for my RPI4B with 4GB.

closed 2019-08-29 by iannecj

pvizeli, 2019-08-29:
Small tipps:
```
hassio ha update
hassio su update
hassio os update
```

This command can you use for update HomeAssistant/Hass.io (aka Supervisor)/HassOS