home-assistant/operating-system #463: 3.4 not booting on RPi 4

dpkay, 2019-08-29:
Hi there,

(First of all, thanks pvizeli for releasing 3.4 today and spending so much time testing – I am looking forward to having the WiFi issue resolved!)

I downloaded 3.4 32-bit today and tried to run it, but my RPi 4 doesn't seem to boot with it. With HDMI connected, the screen stays completely black.

If I insert the 3.3 SD card, I get to the same state as before (WiFi not found), and my Raspbian SD card boots fine too. So I think my RPi and HDMI setup is fine.

Is there any way I can debug this or extract log data? I'm not even getting a blinking cursor or anything. I'm new to the RPi ecosystem, so not sure what my options are.

Thanks!

pvizeli, 2019-08-29:
I have no mini HDMI. I use the RPi4 all time with uart interface on a serial console. That works fine.

Maybe you need add `dtoverlay=vc4-fkms-v3d` to your `config.txt` on RPi4. That is set with raspbian.

rmarkle, 2019-08-29:
Same issue here. HDMI output was working on 3.3

fjfricke, 2019-08-29:
I have a Hifiberry DAC HAT on my RPi4 and can set that as an output in various Hassio Addons. But I don't get any audio output. Is that related to that issue?

dpkay, 2019-08-29:
> Maybe you need add `dtoverlay=vc4-fkms-v3d` to your `config.txt` on RPi4. That is set with raspbian.

Sorry for the ignorance - can you point me at where I create this config.txt?

I haven't really been able to configure things (e.g. WiFi) through files on the flash drive after many attempts and gave up so far, and from other threads it seems like people resolved that by using nmcli on the console (using USB keyboards and video out). I'm still trying to get the initial setup up and running.

reichley, 2019-08-29:
I'm in the same boat...I'm transitioning to hass.io after traditional install for a couple years and can't get my pi4 (4GB) working. I'm plugged into ethernet and get an ip but after 20, 40, 60 minutes no port 8123 access via the browser. I plugged in hdmi to both ports and neither show anything besides blank screen. Also, I've hooked up serial to pins 8 and 10 (and ground) and get no output when booting and using screen command. Any glaring problems why I can't get access via serial or blank hdmi?

pvizeli, 2019-08-30:
I have no hardware to test or fix this issue. I put it to my wishlist for next shoping.

carlostico, 2019-08-30:
Same issue here. HDMI output was working fine on previous release

skynet01, 2019-08-30:
> I have no hardware to test or fix this issue. I put it to my wishlist for next shoping.

where is your wish list i'll buy it for you :)

cogneato, 2019-08-31:
@dpkay 
config.txt is on the SD card in the hassos-boot partition. You may have to use the disk utility in a linux OS or OS X in order to see it and mount it (it's the one FAT partition on the SD)

Normally on Windows it is the one partition that shows up without issue, but I now notice I can not make the partition active. I did access it via a linux pc, but adding `dtoverlay=vc4-fkms-v3d` did not make a change.

m2kw6s, 2019-09-05:
First timer with Hassio.  I started out with the 3.4 image on a RPi4 Model B and was confused by the no HDMI video issue.  Found this GitHub issue about the same program, and I tried the following with success:
used an Rpi2 with a USB SDcard adapter to mount the Rpi4's SDcard with the Hassio image on it.  On the Rpi2: Mounted /dev/sda1 (/boot partition on the Hassio image) to /mnt and then added 
dtoverlay=vc4-fkms-v3d to /mnt/config.txt
umount /mnt, and booted up the Rpi4 with the new config.txt, and I now get video of the boot process/console.

RuBiCK, 2019-09-06:
Same issue here and solved by adding `dtoverlay=vc4-fkms-v3d` to `config.txt`

closed 2019-09-09 by pvizeli

pvizeli, 2019-09-09:
For history, everyone with the problem can solve that like above or should reflash with > 3.4

PR #472 (2019-09-09): Fix screen issue with RPi4
Fix: #463