home-assistant/operating-system #466: Feature Request: usbip support

up2urheadlights, 2019-09-03:
I use hyper-v server to run hassos which does not easily support usb device passthrough to a guest os.  Thus I want to use the linux usbip tool within hassos to mount usb devices that are attached to another linux computer.  Mainly to use a z-wave usb stick.  When support is added to hassos, then an addon could also be developed to configure usbip server address to automatically connect all usb devices on the server.  This could also help if one wanted to run hassos remotely and yet have the z-wave hardware located locally, or vice versa.

usbip is in the linux kernel, so adding a few packages and loading the kernel module should be all that's needed to add support.  I'm hoping one of the developers here will find this a worthy addition and implement it.  Thanks!  

I am aware that this is possible by installing hassio on top of ubuntu, which is how my current setup is, but I'd rather everything be integrated into hassos to use the least amount of resources possible and also for ease of setup.

rbflurry, 2019-09-07:
+1 for this or socat for ser2net support

ocalvo, 2019-09-29:
I am also interested in this. 
@pvizeli I can look into adding this. Will this be something that you or the community will be willing to approve?

pvizeli, 2019-09-30:
https://github.com/home-assistant/architecture/issues/285

It's not so simple as it look like

closed 2019-12-06 by stale[bot]

CleverCompiler, 2020-08-27:
Yes Please.

git-developer, 2022-02-03:
[RUDY](https://github.com/git-developer/rudy) wraps USB/IP in a Docker container, maybe interesting here.