home-assistant/operating-system #469: Apparent CPU overheating issues on Pi 4B after upgrading from 3.3 to 3.4

stimpy2318, 2019-09-06:
Unfortunately, due to the lack of the proper /sys/ entries, I was unable to monitor my CPU temperatures on 3.3 on my Raspberry Pi 4B 4GB. However it was fairly stable and I do not recall it locking up or having significant performance issues after running for several days to a week. 

I recently upgraded to 3.4 shortly after it was released and now every day or two the performance gets to the point of locking up or degrading due to throttling where almost nothing is possible without a physical reboot. After having this issue a couple of times I did some digging and found that the /sys/ entries are now available for cpu temp in 3.4 and I setup a sensor for it. I immediately noticed that the temperature was dangerously close to the 80C throttle temp once the sensor was setup.

No physical changes to the device were made during the time when it was stable on 3.3 and when it started having lockup issues on 3.4. It was kept in the same location at the same temperatures and in the same case with the same heat sinks. No new addons were loaded and the cpu load and memory has remained fairly consistent.

I am not an expert with Raspberry Pi, but looking over the release notes for HassOS and doing a bit of research, it may seem that a somewhat faulty version of the Pi firmware was used in 3.4.

In order to keep things operational for now, I have since discovering the overheating, decided to keep the top off the case that the Pi is in while I work on an automatically switched cooling fan setup for it. However this probably shouldn't be necessary since a Pi (even a 4) with a decent heat sink setup should probably be able to maintain low CPU load levels (mine hangs around 5-17%) without the need for active cooling.

pvizeli, 2019-09-06:
We follow up the original firmware and this version is with the optimized one. Anyway I update the firmware again before the stable branches would merged. That is what I can do

stimpy2318, 2019-09-07:
Understood, I just wanted to make sure that the behavior that I saw was on your radar in case others are experiencing the same issues.

As for a somewhat related question, once 3.x is in the stable branch, what is the process for switching to use the stable branch going forward instead of development branch?

Mariusthvdb, 2019-09-19:
can only confirm a rather warm Pi4, even in practical idle state:

[image]

this is a quick history, reaching almost 80 degrees..

[image]

pvizeli, 2019-09-20:
Is latest firmware there. Anyway, you need a cooler for you RPi4, the firmware can help a bit but they will be around 5-10c°

Mariusthvdb, 2019-09-20:
> Is latest firmware there. Anyway, you need a cooler for you RPi4, the firmware can help a bit but they will be around 5-10c°

Firmware: you mean Hassos? btw can I update to rc 3.5 using the hassio/system/ update button? I am on 3.4

and, will the firmware make the 5-10% difference, or is the cooler needed for that ...

ocalvo, 2019-09-29:
How do you get the system statistics, is there a component for that?

closed 2019-12-06 by stale[bot]