home-assistant/operating-system #473: Does not request IP from DHCP.

sbeaudoin, 2019-09-12:
- Created generation 2 hyper-v vm with hassos_ova-2.12.vhdx, 1GB static RAM (as in "not dynamic").  
- Added the vm MAC to my DHCP server to provide a static address.  
- Boot ok, but does not request any address to the DHCP server, stuck at "udevd[8]: starting eudev-3.2.8" after 7 seconds.  Hyper-V console show no ip address.
- I can see in the boot screen "IPv6 ... link becomes ready", but nothing about IPv4.
- No IP, so, no access to the web UI.

Any suggestions?

tomlut, 2019-09-29:
Same problem with the NUC image. https://github.com/home-assistant/hassio/issues/1310

LukasQ, 2019-10-03:
Which vSwitch do you use? Which Hypervisor?
Could you try to disable the MAC Filter on your DHCP Server?

tomlut, 2019-10-03:
Hypervisor: N/A (NUC image).
Directly into and Edgerouter X.
MAC filtering is not enabled.

Works fine when using the docker install method on Ubuntu Server LTS.

sbeaudoin, 2019-10-03:
The vSwitch is configured to be an "external" switch and it is shared with the host.  The hypervisor, as I said, is Hyper-V on a Windows Server 2016 Standard host.  When you say "disable the MAC filter", I suppose you want the VM virtual NIC to get a random IP from the DHCP server and not a static one associated with the MAC.  Or, do you want me to assign a static IP on the card?  I suppose it is the first one because I can't assign a static IP on the card when the machine does not boot.  I don't understand how a DHCP filter would cause this problem, but I will try anyway,

sbeaudoin, 2019-10-03:
Removed the MAC adress association from my DHCP server.  Same symptoms.  More details : The IPv6 link state toggle from ready to not ready.  Still nothing about IPv4 in the console and no IP adress shown on the HyperVisor.  

If I look on the DHCP server for a lease for this MAC, I can see 10.0.20.239 though, but no connection with "http://10.0.20.239" or "https://10.0.20.239" from my own PC.

Here is a screenshot if it can help you.

[image]

LukasQ, 2019-10-04:
Sorry, yes MAC association. Which Configuration Version do you use? 8?
I had basically the same problem on my Hyper-V Server 2016. Try to re-download the image and recreate a new VM. Then, check Secure Boot Options. I know there's a Bug in Server 2016 with it.
May sounds confusing, but that worked for me.

Plus, I read something about Docker on Hyper-V not passing the correct option sets on nested virtualisation. Thus ´´Set-VMProcessor -VMName "<Your Virtual Machine Name>" -ExposeVirtualizationExtensions $true´´

sbeaudoin, 2019-10-04:
Yes, it is a version 8.0.  The last command you give me look like powershell, is it?  When you say "Check secure boot options", what options do you want me to check?  Presently, none of the checkbox is checked on any of the three group box.  I don't understand why recreating the same VM with the same image should give a different result, but I will try.

sbeaudoin, 2019-10-04:
Before reinstalling the VM, I executed the command on my present VM.  No changes.  I will try again on the new VM.

sbeaudoin, 2019-10-04:
Tried a new VM with the new OVA 3.5 image, same symptoms.  As I specified before, I was using OVA 2.12 before.  Execution of the powershell command afterward did nothing.  I understand you have a working hyper-v VM with an OVA image?  If so, is it possible to provide me your hyper-v VM files (without the vHardDisk) so I can "mount it", try it and if it works, compare this one and mine?

LukasQ, 2019-10-04:
So, one after another.
In my case, I think I remember doing the same thing again solved my problem. It's a miracle, as Queen would say and often the case with MS products IMHO.
Did you try the vhdx or the ova? 
Unfortunately I use 2019 with Config v9, so my VM wouldn't run on yours.
You have to execute the powershell command on the Hypervisor
What if you run Hyper-V on your computer? (If it's WIn10, then you could use my machine)

sbeaudoin, 2019-10-05:
I tried the only image who, I understand, will work on hyper-v : hassos_ova-3.5.vhdx.gz.  As explained, I did execute the command on the hypervisor (host).  Even if I try and run it on Win10 and works, it is not what I need and will not help to make it work on Server 2016.  IMHO, with linux, the simplest thing involve modifying obscure text files somewhere you must google to find and where the content have cryptic single words parameters with difficult to find help for their purpose and the possible values.  Or you must download tools not pre-installed that a noob like me must search to find which and spend a whole day for something who should take 5 minutes, like this one.   The reason I want a pre-built image is because I didn't want to lose time to configure it.  Plug and play, I thought.  I kn ow network, I know VM, so, how hard should it be!  Well, this is another "easy" linux thing who become almost impossible to solve without a lot of time.  I declare forfeit, I will wait for a Windows version, event if it takes years.

LukasQ, 2019-10-06:
@sbeaudoin I downloaded the official Hyper-V image from here: https://github.com/home-assistant/hassos/releases/download/2.12/hassos_ova-2.12.vhdx.gz, 'cause, as written in the documentation, the OVA won't work atm.
Then I, as I'm somehow are in a good mood ;) , created an older Hyper-V VM for you here: 
https://internal-hass-dev.s3.eu-central-1.amazonaws.com/HassOS-8.rar
1) Download the RAR, 
2) extract it to your Hypervisor,
3) Replace the "hassos_ova-2.12.vhdx" Folder with the image above
4) Import it in the Hyper-V MMC to your Hypervisor
5) Go and learn Linux. You'll need it in the future. #WSL

If this is still not working, you try Docker on Windows.

sbeaudoin, 2019-10-06:
Thank you for the mood and effort, but I still have the same problem.  I now suspect an incompatibility with my network card.  I got a true Intel 1Gb card for cheap, I will try it some times in the future and it will surely be better than my on-board realtek card.  I'll need linux in the future?  Yeah, right...  At the same time I will appreciate a good punch in the face!  Who knows...  ;-)

closed 2019-12-12 by stale[bot]

jasonhyland, 2020-01-22:
I have the exact same issue, no IP address issued. Looks like it wasn't fixed since reported back in Oct.

g333k, 2020-11-15:
I have the exact same issue too, not connecting to ipv4 supporting router. Same works fine with ipv6 supporting router. Looks like it still isn't fixed since reported back in Oct.

erandermsft, 2020-12-13:
Pretty sure it's related to the WIFI Adapter (or driver). Had the exact same issue trying to run my WIFI Adapter in bridged mode in Hyper-V. Resolved by bridging the Ethernet adapter and using a wired connection instead, with no other changes.