home-assistant/operating-system #480: No network on hassos 3.5 with dtoverlay=pi3-disable-bt

olbjan, 2019-09-14:
I tried the update to hassos 3.5 from 2.12 on my Raspberry Pi 3B and the update ended with the network not being available anymore.

I have narrowed this down to having
```
dtoverlay=pi3-disable-bt
```
in my /boot/config.txt file to use the my Razberry Z-Wave board. This parameter was necessary to have the Z-Wave controller module available and is documented here:
https://www.home-assistant.io/docs/z-wave/device-specific/#razberry-board

Without the dtoverlay=pi3-disable-bt in config.txt my Pi is booting properly (but I suppose won't have zwave anymore.

pvizeli, 2019-09-14:
you need only `enable_uart=1`

olbjan, 2019-09-14:
With ```enable_uart=1``` in config.txt on 3.5 HA isn't seeing the Razberry board. The network is coming  up though.

pvizeli, 2019-09-16:
Yeah, look there is an issue with RPi official 4.19 linux kernel

Try2Fly, 2019-09-16:
ok, only entering `enable_uart=1` also bricks my system.
I'm using hassOS 2.12 btw, I do not have the option to update to 3.5

Nemesis24, 2019-09-16:
It's the same for me. with `dtoverlay=pi3-disable-bt` I can't get wifi, without i can get wifi but i can't get my raspbee working.

Nemesis24, 2019-09-16:
In fact for ZHA you must change your integration of the device to` /dev/ttyS0 `and not `/dev/ttyAMA0`
It works perfectly for me.

Try2Fly, 2019-09-17:
I’m also trying to install the raspbee. According to hassio the raspbee is installeer on ttyAMA0.

Also reverting back nu deleting the entry again does not help. System stays bricked. I have to format and reinstall everything again. Is there a way to get it working again? Maybe by editing or deleting some files?

Nemesis24, 2019-09-17:
Have you try to change` /dev/ttyAMA0` in `/dev/ttyS0` ?

Try2Fly, 2019-09-17:
No I did not try that. So you suggest to only change to `/dev/ttyS0` and not change the config.txt file?

Nemesis24, 2019-09-17:
In fact I suggest :

1.  Delete` dtoverlay=pi3-disable-bt` in config.txt
2.  Change in your configuration of rasbee integration` /dev/ttyAMA0` in `/dev/ttyS0`

Try2Fly, 2019-09-17:
The web UI (phoscon) won't start, and in the log of the deconz addon it it shows: `No device /dev/ttyS0 found!`

When I use /dev/ttyAMA0 the web UI does start, but there is says the Raspbee is not connected

Nemesis24, 2019-09-17:
I don't know for you but for me all works fine like that.

Try2Fly, 2019-09-18:
Ok, just to be sure. I did not add `dtoverlay=pi3-disable-bt` so I cannot delete. I also did not add `enable_uart=1` to the config.txt file. 
both  `/dev/ttyAMA0` and `/dev/ttyS0` do not give me a working raspbee.
For `/dev/ttyS0` the add-on won't start
For `/dev/ttyAMA0` the addon starts, but Phoscon states that the raspbee is not connected.

Nemesis24, 2019-09-18:
Ok for me i've got  `enable_uart=1` in the config.txt file, and i delete `dtoverlay=pi3-disable-bt`.
In Zha Configuration I change `/dev/ttyAMA0` in `/dev/ttyS0`, and i've got my raspbee working fine.

I found this if it could help
[(https://community.home-assistant.io/t/raspbee-raspberrypi-4/132547/8)]

olbjan, 2019-09-18:
I now also have a Pi4 to play around with and neither
```enable_uart=1``` or ```dtoverlay=disable-bt``` or any combination thereof got me the RazBerry available on either /dev/ttyAMA0 or /dev/ttyS0
This was tested on the 3.5 hassos 64-bit image.

Nemesis24, 2019-09-19:
My combination is with a raspberry 3B+ ... i put my raspberry on 5g wifi because i've got a router who could do 2.4 and 5g, and with 5g it's more stable and i deactivate MIMO function.
for the rest i'm on hassos 3.5 with a raspbee who works on /dev/ttyS0 with enable_uart=1 in config.txt file.
I don't know more i'm really sorry.

pvizeli, 2019-09-19:
The issue is that: https://github.com/agherzan/meta-raspberrypi/issues/374

I have it high on my task list. I hope to work on it next week.

Markuzzo, 2019-10-30:
On my Rpi3 B+ with Hassos 3.5 and raspbee, I solved with:
1. config.txt:   only the line: enable_uart=1
2. DeConz config: /dev/ttyS0

PR #503 (2019-12-10): Fix RPI issues with u-boot / firmware
Fix: #499
Fix: #492
Fix: #488 
Fix: #480
closed 2019-12-10 by pvizeli