home-assistant/operating-system #481: Enable Intel bluetooth on ova image

ETaube, 2019-09-15:
This weekend I moved my HA setup from a PI3 to an Intel-nuc(NUC7CJYH). I installed proxmox6 on the nuc and created a VM with the hassos ova image. Everything works fine expect for the bluetooth module. The USB bluetooth module is detected by the OS, checked it via lsusb command, but I guess the drivers are missing. 

Just created another vm with the hassos nuc image. With this image I had to change the network adaptor from VirtIO to Intel e1000 in proxmox to get the network working. But in this case the bluetooth device is working.

Is it possible to add intel bluetooth support to the ova image? Since it is the way to go when running hassos in a virtual environment.

SilverWx, 2019-09-19:
I am trying to use both CSR and BCM Bluetooth dongle without succeed in VMware, it seemed that the driver is missing here also.

vooon, 2019-10-19:
Same there. I'm trying to use CSR dongle with VMDK image running by bare qemu (with kvm).

qoobaa, 2019-12-01:
I've got two different dongles and none of them seems to work in HassOS. I suspect the kernel modules are missing. Is it possible to include them?

qoobaa, 2019-12-01:
Is there any dongle out there, that actually works with the current HassOS?

pholausson, 2019-12-03:
Same here, tried internal bluetooth on intel nuc and 2 different dongles , no luck getting bluetooth started in hassio on hassos nuc image

Jopinder, 2019-12-04:
Same issue with Unraid on ASRock H370M-ITXac. I'm passing through the onboard usb bluetooth (Intel Corp. (8087:0aa7)) to the VM and I can see it with lsusb (Bus 003 Device 002: ID 8087:0aa7) but still unable to make HassOS/HA recognize the adapter.

elmurato, 2020-01-07:
You can add me to the list. Running Hass.io OVA in Proxmox VE 6.1 on Intel NUC6CAYH. Passed USB device 8087:0aa7 in PVE to Hass.io, which is afterwards visible in Hass.io via 'lsusb':

> Bus 002 Device 002: ID 8087:0aa7

'dmesg' shows only following message:
> usb 2-1: New USB device found, idVendor=8087, idProduct=0aa7, bcdDevice= 0.01

After entering 'bluetoothctl' and hitting enter there is no output at all. The bluetooth device hci0 is not created.

It would be great if bluetooth support would be added to the OVA version as this is the only thing which is not working on my system. Unfortunately I need it for my bluetooth thermostats :( 

Please let me know if/how I can help or support to fix this issue. Thanks in advance.

ezaton, 2020-01-22:
I have to add my comment on this as well. I am using KVM (with some later patches) on my hardware, and the OVA image, after slight format modifications works like charm. However - I can pass-through BT device to the system, and I do see it under 'lsusb' but there are not BT drivers compiled to this kernel. Is my only option is to recompile the kernel with "my" settings?

Twanislas, 2020-02-05:
FYI, I believe this could be fixed by #563 (dev build 4.1 includes the fix).

elmurato, 2020-02-11:
Home Assistant OS 3.10 fixes the missing bluetooth support. My bluetooth thermostats are running again with HA (OVA image) in Proxmox on an Intel NUC. Awesome! Thanks so much @ezaton and @pvizeli 🎉

closed 2020-02-11 by pvizeli