home-assistant/operating-system #482: add support for qemu-guest-agent

JoeyG1973, 2019-09-16:
Please add support in hassio for the qemu-guest-agent

https://wiki.qemu.org/Features/GuestAgent

buginside, 2019-09-29:
I use hassos in Proxmox VE, qemu-guest-agent could be used for:

1-Properly shutdown the guest, instead of relying on ACPI commands
2-Freeze the guest file system when making a backup

pvizeli, 2019-09-30:
I don't block this, make sure that the guest agent is only started on the correct virtualisation like vmware tools.

JoeyG1973, 2019-10-01:
@pvizeli I am not sure what you are trying to say.

There doesn't seem to be a straight forward way to install qemu-guest-agent or anything to hassos. This can't be installed as a container.

pvizeli, 2019-10-01:
It's buildroot on that repository, not containers. You need learn how buildroot work and you are able to create an PR with this feature. You can look at it as hacktoberfest chellange

mihalski, 2019-10-18:
I searched to find how it is possible to determine the virtualisation technology being used and found:

https://unix.stackexchange.com/questions/89714/easy-way-to-determine-virtualization-technology

It seems dmidecode would be a really useful command to determine the virtualisation technology being used:

```
dmidecode -s system-product-name
VMware Virtual Platform

dmidecode -s system-product-name
VirtualBox

root@router:~# dmidecode -s system-product-name
KVM
```
Would it be reasonable to include it in Hassos for this purpose?

Regards,
Michal

pvizeli, 2019-10-18:
Because we use systemd, you can easy say which service should start on which virtualization. We do that for vmware tools.

However, I do not add or work on this feature. If someone implement it and make a PR, I'm fine to merge this.

mihalski, 2019-10-19:
That's way over my head for now.

I was not able to figure out how VMware was detected and then I couldn't figure out how to build (or if it's possible) the qemu-guest-agent without also building the entirety of Qemu.

Maybe someone else is a bit more capable? I should be able to test patches if that helps.

pvizeli, 2019-10-21:
https://buildroot.org/downloads/manual/manual.html

We simplify some steps with make script and docker but to contribute to this project, you need understand how buildroot work. For small things like detect the virtualisation, I can help you on the PR.

jens-maus, 2019-12-06:
FYI: As I am maintaining a similar buildroot-based operating system project (see https://github.com/jens-maus/RaspberryMatic) and just introduced virtual/ova support in the latest version, I also stumbled over the missing qemu-guest-agent in buildroot and just developed an adequate buildroot package myself.

See here for more information:
https://github.com/jens-maus/RaspberryMatic/tree/master/buildroot-external/package/qemu-guest-agent

Feel free to grab that for your own purposes here in hass-os

pvizeli, 2019-12-09:
@jens-maus cool, thanks

closed 2019-12-12 by pvizeli

timdonovanuk, 2020-02-07:
Sorry, I'm really dumb when it comes to tracing github merges. Is this supposed to now be in HassOS? I'm on HassOS 3.9 with the beta Supervisor (199) and I still get an error about qemu guest agent not starting when I watch HassOS load up. Cheers!

moeffju, 2020-02-08:
For everyone else struggling with this - make sure to enable the Qemu agent support on the host node first so that the appropriate device is created in the guest. You likely need to shut down and restart the guest. Then the qemu-ga should run fine in the guest.

ZoidbergZulu, 2021-05-20:
I apologize for reviving this old thread. I'm running hassos (or HAOS) on KVM/QEMU and my backup plans depend on the availability of QEMU Guest Agent. 
I have configured the guest agent on the host side and it's working fine with all other VMs. However, I cannot connect to hassos from the host system as there appear to be no binaries on hassos for qemu-qa.
A search for qemu on hassos reveals the following results:

`
~ $ find / -iname *qemu*
/dev/virtio-ports/org.qemu.guest_agent.0
/usr/share/bash-completion/completions/qemu-system-x86_64
/usr/share/bash-completion/completions/qemu-system-i386
/usr/share/bash-completion/completions/qemu
/usr/share/bash-completion/completions/qemu-kvm
/sys/devices/pci0000:00/QEMU0002:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/QEMU0002:00
/sys/bus/platform/devices/QEMU0002:00
/sys/bus/acpi/devices/QEMU0002:00
`

The qemu device seem to be available, however binaries are missing.

Does anyone have an idea how to (manually) fix this? I'm not backing up my homeassistant instance which I find a bit scary.

Thanks much in advance for any hint. I've spent already hours searching for a solution.

jens-maus, 2021-05-20:
@ZoidbergZulu According to https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/package/qemu-guest-agent/qemu-guest.service#L8 the `qemu-ga` binary should be available in `/usr/libexec/`

ZoidbergZulu, 2021-05-20:
Thanks for the swift response. That directory is almost empty, I only see a host of git-related binaries and dbus-daemon-launch-helper. 
Could anyone check how their libexec directory looks like?

~ $ cd /usr/libexec/
libexec $ ls
dbus-daemon-launch-helper  git-core

I'm running Home Assistant OS 6.0.rc2.

jens-maus, 2021-05-20:
> Could anyone check how their libexec directory looks like?

Of course, see here:
[image]

> I'm running Home Assistant OS 6.0.rc2.

You didn't tell which base image you use for your KVM/QEMU host. Here I simply imported the ova build in my ESXi and started the host and voila, the necessary binaries are there. Thus, note that the `qemu-ga` is only available for the ova platform and not available for rpi or the other arm-based platforms.

ZoidbergZulu, 2021-05-20:
I believe I have used the correct image but maybe one of the other conditions are not met.

This my KVM configuration stub (platform-specific details). HyperV is enabled as well.

<domain type='kvm'>
  <name>hassos</name>
  <metadata>
    <libosinfo:libosinfo xmlns:libosinfo="http://libosinfo.org/xmlns/libvirt/domain/1.0">
      <libosinfo:os id="http://redhat.com/rhel/8.1"/>
    </libosinfo:libosinfo>
  </metadata>
  <memory unit='KiB'>4194304</memory>
  <currentMemory unit='KiB'>2097152</currentMemory>
  <vcpu placement='static'>4</vcpu>
  <os>
    <type arch='x86_64' machine='pc-q35-5.2'>hvm</type>
    <loader readonly='yes' type='pflash'>/usr/share/qemu/edk2-x86_64-secure-code.fd</loader>
    <nvram>/var/lib/libvirt/qemu/nvram/hassos_VARS.fd</nvram>
    <boot dev='hd'/>
  </os>
  <features>
    <acpi/>
    <apic/>
    <hyperv>
      <relaxed state='on'/>
      <vapic state='on'/>
      <spinlocks state='on' retries='8191'/>
      <vpindex state='on'/>
      <synic state='on'/>
      <stimer state='on'/>
      <reset state='on'/>
    </hyperv>
  </features>
  <cpu mode='host-passthrough' check='none' migratable='on'>
    <cache mode='passthrough'/>
    <feature policy='require' name='topoext'/>
  </cpu>
  <clock offset='localtime'>
    <timer name='rtc' tickpolicy='catchup'/>
    <timer name='pit' tickpolicy='delay'/>
    <timer name='hpet' present='no'/>
    <timer name='hypervclock' present='yes'/>
  </clock>

Thanks for any ideas here.

ZoidbergZulu, 2021-05-20:
Small update from my side: I could solve the issue by re-creating a new VM with a fresh config. First, I was able to replicate that the service is indeed running, at least I could talk to the agent via the host system. I copied my old VM image into the newly created system and voila I could still interact with the agent on the guest system from my host.

The only strange thing is that there is still neither a qemu-ga binary visible on the filesystem nor a process running. I'm not familiar with the architecture of hassos but could it be that I'm not seeing the basic OS but rather a container running on top of it?

ZoidbergZulu, 2021-05-20:
While I'm still puzzled as to why I cannot see the process or binaries of qemu-ga, I think I have found the culprit.

When setting up my VM in the first place, I have copied over the HyperV enlightenments settings from my Windows 10 VM (I was not sure if they add much value to performance but I believed they couldn't hurt either). Removing these lines from my config did the trick.

Thanks much for your support, @jens-maus!

bpeperkamp, 2024-12-29:
> While I'm still puzzled as to why I cannot see the process or binaries of qemu-ga, I think I have found the culprit.
>
> When setting up my VM in the first place, I have copied over the HyperV [...]

Can confirm that this was the culprit ;-)

PR #505 (2019-12-12): Add qemu guest agent
Fix: #482