home-assistant/operating-system #489: esxi , no usb visible, passthrough not possible, doesnt boot

pergolafabio, 2019-10-09:
Guys

i run esxi server with hassos 2.12

i bought this video to usb device , for my doorbell automation

https://www.amazon.com/Usb2-0-Capture-Compatible-Window-Black/dp/B00RMYWGWC/ref=sr_1_3?keywords=composite+capture+card+for+pi&qid=1555039000&s=electronics&sr=1-3-catcorr

first try, i just added the USB device, like i do with my conbee stick
But Hassos , in hardware tab, i didnt see the device
so tried with ubuntu & Windows, then i saw the device and also /dev/video0
but for some reason, i think that device is not compatible with the USB virtual controller of esxi, cause there was no video playback

second try, then in esxi i did a passthrough of my complete usb 3.0 controller, now in windows & ubuntu it works... yey!

but if i passthrough the usb 3.0 controller and attach it to my hassos VM, my hassos doesnt start, it doesnt even show the vmware bios log, so its just doesnt start... i have no idea why?

anyone some tips?

closed 2019-12-15 by stale[bot]

narsaw, 2020-02-16:
> Guys
>
> i run esxi server with hassos 2.12
>
> i bought this video to usb device , for my doorbell automation [...]

Hi, did you ever figure this out? I am trying to passthrough a USB device to a hassos VM but it will not boot.

pergolafabio, 2020-02-16:
Nope, I setup a second host with that device to capture the stream

narsaw, 2020-02-16:
Thx I’ll keep looking

pergolafabio, 2020-02-16:
Keep me posted ;)

RedNax67, 2021-11-14:
A year and a half later this is still not fixed. Moved to a supervised installation on ubuntu. Everything works fine now.

agners, 2021-11-15:
@RedNax67 it has been closed because there was no activity on the issue for a while.

Now this issue is about OS 2.12, at this point more than 2 years old. If you can reproduce it with your setup, with a more recent OS version, please open a new issue.

Yiivgeny, 2022-01-07:
Problem still on Hassos 7.1 with any passthrough device (wifi card in my case)

agners, 2022-01-07:
@Yiivgeny as pointed above, can you please open a new issue then? Also please provide which ESXi version you are using, what type of WiFi card you are using (including how the card is connected, which Bus/Type).