home-assistant/operating-system #490: Impossible install of hassos on 32gb on raspberry 3b+

Nemesis24, 2019-10-19:
Ok i will try to explain... i've got using hassos 3.5 on a sdcard 32gb on a raspberry 3b+ and after a reboot, my install was broken.

For the first time, i think that my sdcard was broken...corrupt
So i decided to make a new install of hass.io on my sdcard 32gb, but the install don't boot.
So i decided to have a new sdcard 32 gb to be sure, but the install don't boot.
Finally i try with an old sdcard in 16gb and the install boot

Thanks

ciotlosm, 2019-10-22:
@Nemesis24 try connecting the rpi 3b+ to a monitor and look at output to understand where it failed to boot.

Nemesis24, 2019-10-22:
I have connected my rpi3b+ on a monitor, but i couldn't see anything because, there was only 20 lines who go very very fast and reboot all the time , on my 32 gb card i can install a rasbian lite without any problems.
For now i use hassio on my 16gb card.

ciotlosm, 2019-10-22:
Which version of hassos did you try? The 32bit or 64bit?

Nemesis24, 2019-10-22:
The 32 bit because recomanded.

cogneato, 2019-10-23:
You can hook up a keyboard and hit enter to interrupt the running log. But it's possible that what you are seeing is the normal log which doesn't really end.

Logging in as root, no password, will get you to the Hassio CLI.

Nemesis24, 2019-10-24:
i've got the keyboard but hit enter do nothing in this case, and it doesn't explain the difference beetween 32 and 16 gb of comportement.

ciotlosm, 2019-10-24:
I tried a 32GB card and it booted for me. Can you try the 2.x version?

Nemesis24, 2019-10-27:
Thanks, but i try again, with 3.5 and 2.12, that was the same thing. I've got only the red led on pi now with 32gb and 16gb perhaps it's related to a bad alimentation, i will try another alimentation... i've got a raspbee, i try without it was the same thing.

ciotlosm, 2019-10-28:
What specs is your current power supply?

Nemesis24, 2019-10-29:
This is a 5v , 2.4A.

caesar1987sk, 2019-10-31:
i have same problem with my 32GB card. infinite loops. But i dont have smaller one :(
Raspbian is running on another 32GB card without problems.

caesar1987sk, 2019-10-31:
its look like hassos_rpi3-1.13.img.gz is working without any problem.

caesar1987sk, 2019-11-01:
after update to latest version (from hassio ui) is again not possible to start it.

MuadDev, 2019-11-03:
I see observe the same behaviour as @caesar1987sk and @Nemesis24 do. I also use a 32GB SD card and a Pi3+.

JokerQyou, 2019-12-01:
Same behavior here, except that I don't see anything on the HDMI monitor, no boot log, nothing. Pi 3 B+ with a 128GB SD card, flashed `hassos_rpi3-2.12.img.gz` with balenaEtcher `1.5.64`.

Matthijsy, 2019-12-10:
I also have the same behaviour on a Raspberry Pi3B. I managed to capture the error, it is complaining about the MBR header. 

Here is the error log (not that clear since it goes very fast)
[image]

Matthijsy, 2020-02-10:
I am not sure this problem is solved already. Could someone give an update?

closed 2020-04-18 by stale[bot]