home-assistant/operating-system #492: some development releases not working on Raspberry Pi 2

lpt2007, 2019-10-29:
Hi,

I wanna let you know that I tested images of those releases on rpi2 and they are not working.

HassOS Release-3 build 5 (RC)
Development 3 build 4
Development 3 build 3
Development 3 build 2

YoranSys, 2019-10-29:
Hi,
Thé same for me,
 HassOS Release-3 build 5 (RC) on raspberry 2 b it's broken

antoniomolram, 2019-11-16:
Hi, for me exactly the same. I tried 
hassos_rpi2-3.5.img.gz
hassos_rpi2-2.12.img.gz

The Raspberry pi the flashing procedure says that CRC are not matching...

antoniomolram, 2019-11-27:
the images are working at all on a rpi2? Could someone try it? I tried raspbian and was working without problem!

QbaF, 2019-11-27:
3.5 not working for me (but only on SSD tested)
2.12 works (on SD)

closed 2019-12-10 by pvizeli

QbaF, 2019-12-12:
on 3.6 RPi2 1.1 still do not boot

pvizeli, 2019-12-13:
Version 3.6 with RPi2 rev. 1.1 work perfect. I know that osX/Mac has some issues and can destroy the boot partition. I would recommend using Linux or windows to flush the SD card.

QbaF, 2019-12-13:
Yes, but You mention SD card. I am trying to boot from SSD. Using Windows, balenaEtcher fo flash. LAN, no WIFi setup, of course, so the most basic procedure. Maybe any important additional steps needed? Preformatting? I will be delighted to test anything what could be working.

PR #503 (2019-12-10): Fix RPI issues with u-boot / firmware
Fix: #499
Fix: #492
Fix: #488 
Fix: #480