home-assistant/operating-system #493: No notification on new HassOS release on Hassio.

sblantipodi, 2019-11-02:
I was running the latest 0.101 hassio on my Pi4 image on HassOS 3.4
Reimaged the microSD with a fresh new image and I noticed that I have HassOS 3.5 now.

There was no notification of the new HassOS on my Home assistant.
Please fix the hassos notification on new released version.

Thanks.

Joeboyc2, 2019-11-02:
I think 3.5 is still in a beta or release candidate stage so you will not
get notification unless you are in the beta ring

sblantipodi, 2019-11-02:
@Joeboyc2 if it's in beta, why the official image uses that version?

Joeboyc2, 2019-11-02:
https://github.com/home-assistant/hassos/releases shows the the 3.5 release
is a release candidate (close to stable, but not a stable release) so to
see that in the interface you need to be on the beta releases.

It's not an ideal situation but thats how it works with the releases

DeliriousMetro, 2019-12-13:
I had joined the beta channel and still could not receive the latest 3.6 release, instead, I had to format a flash drive labelled CONFIG and then place the HassOS release .raucb file on the root directory.

[image]

After doing so, I can see that it has updated to HassOS 3.6 after a couple of minutes but the web interface did need some time to catch up after the boot sequence - but there seems to be the update button now but it just tries to grab HassOS 3.5 again - it does fail with error 500.

[image]

sblantipodi, 2019-12-13:
+1
HA never notified a new hassos release since ever.

the only way to update hassos is to reflash the micro sd with a new image.

please fix.

pvizeli, 2019-12-14:
3.6 is out and @sblantipodi is complete wrong.

`hassio os update --version 3.6` or `hassio os update --version 2.12`

closed 2020-02-13 by pvizeli