home-assistant/operating-system #494: Please add the possibility to move the recorder to external USB SSD.

sblantipodi, 2019-11-02:
As title.
Recording a lot of sensors on microSD burns it fast.

Please add the possibility to move the recorder to external USB SSD.

poudenes, 2019-11-26:
Maybe possible to use mysql and store the dB on a different storage?

Joeboyc2, 2019-11-26:
Have you checked out the influxdb addon, that could probably help you out,
or take a look the documentation for the recorder component to tune it
outout

sblantipodi, 2019-11-26:
> Maybe possible to use mysql and store the dB on a different storage?

this requires a NAS or something similar with really low latency access, 
USB disk is an easyer/cheaper solution.

pvizeli, 2019-11-27:
You can setup a database on a database server and use that. Please no feature requests as PR.

closed 2019-11-27 by pvizeli

sblantipodi, 2019-11-27:
> You can setup a database on a database server and use that. Please no feature requests as PR.

@pvizeli it is not possible with hassio that is the only reason why hassos exist.
hassio can't use a VPN for this, no VPN, no way to setup a database server, I mean, I don't think that someone want to expose a DB over the internet.