home-assistant/operating-system #495: Update

Monacoslo, 2019-11-08:
Update button on Configuration pane installs earlier version. If you are using hassos on ssd(usb), you can not access your system anymore since on nevely installed version boot from usb is not supported

cannotdraw, 2019-12-03:
same issue. i just accidentally updated to 2.12 from 3.5 =[

knowsnotmuch, 2019-12-05:
> same issue. i just accidentally updated to 2.12 from 3.5 =[

Yes - flashed 3.5 then saw the "Update" button - clicked it and boom 2.12 was downloading :(

Joeboyc2, 2019-12-05:
You need to join the beta channel to stay on the 3.5 branch, it is not out
of release candidate yet

knowsnotmuch, 2019-12-05:
> You need to join the beta channel to stay on the 3.5 branch, it is not out of release candidate yet
> […](#)

Did join the beta channel. Still downloaded 2.12 when I clicked update.

knowsnotmuch, 2019-12-13:
Just an update. I tried by joining beta channel, RELOADING supervisor and then checked and I stayed on 3.5. I am not sure if reloading addresses this.

Monacoslo, 2019-12-13:
> You need to join the beta channel to stay on the 3.5 branch, it is not out of release candidate yet
> […](#)

Where to join?

bbbenji, 2019-12-31:
Is there a known simple way to get back up and running if accidentally downgraded? I don't want to reinstall and configure HassOS and restore a snapshot... But looks like that's what I will have to do.

knowsnotmuch, 2019-12-31:
> Is there a known simple way to get back up and running if accidentally downgraded? I don't want to reinstall and configure HassOS and restore a snapshot... But looks like that's what I will have to do.

Not sure @bbbenji - I went down the reinstall path, but this _could_ work from the command line 
`hassio hassos update --version=x.x`

bbbenji, 2019-12-31:
Thanks @knowsnotmuch 

I ended up flashing the latest HassOS to the SD and it booted right back into Home Assistant. Didn't have to restore anything.

Monacoslo, 2020-01-01:
Since the latest hass.io official version is 3.7 after update everything should work ok

closed 2020-03-08 by stale[bot]