home-assistant/operating-system #496: Add support to CEC on Odroid c2

tomszcz-web, 2019-11-10:
I'm using Home Assistant in docker on Odroid c2 (homeassistant/odroid-c2-homeassistant).

Everything works great except CEC which does not work:

```
bash-5.0# cec-client -l
libCEC version: 4.0.4, git revision: libcec-4.0.4, compiled on Thu Nov  7 16:26:24 UTC 2019 by root@5be64fb631de on Linux 4.15.0-1061-azure (aarch64), features: P8_USB, DRM, P8_detect
Found devices: NONE
```

To support cec in Odroid c2  libcec have to be compiled with passed argument -DHAVE_AOCEC_API=1 to the cmake command in the compilation instructions.

If it is possible please add this argument in compile instructions.

Thanks.

pvizeli, 2020-01-10:
Yes. Looking at hassio-homeassistant repository as example the rpi3

Twanislas, 2020-02-05:
Hi !
FYI, I believe this could be fixed with #565 (in dev build 4.1 for the moment).

pvizeli, 2020-02-06:
Yes and need also fix later on the image itself

closed 2020-04-14 by stale[bot]