home-assistant/operating-system #497: VMDK won't snapshot or clone

datdamnzotz, 2019-11-11:
There *IS* something funky with distributions of the the vmdk

Snap shoting or attempting to clone the VM from disk image https://github.com/home-assistant/hassos/releases/download/2.11/hassos_ova-2.11.vmdk.gz
or
https://github.com/home-assistant/hassos/releases/download/3.5/hassos_ova-3.5.vmdk.gz
(the two I tried)  

Fails with **Object type requires hosted I/O**

In Esxi 6.7 the Guest log file:
2019-11-11T03:39:17.573Z| vmx| A100: ConfigDB: Setting ide0:1.redo = ""
2019-11-11T03:39:17.573Z| vmx| I125: DISK: OPEN ide0:1 '/vmfs/volumes/86cb31ac-0843a890/Home/hassos_ova-3.5.vmdk' persistent R[]
2019-11-11T03:39:17.578Z| vmx| I125: DISKLIB-LINK  : "/vmfs/volumes/86cb31ac-0843a890/Home/hassos_ova-3.5.vmdk" : failed to open (Object type requires hosted I/O).  
2019-11-11T03:39:17.578Z| vmx| I125: DISKLIB-CHAIN : "/vmfs/volumes/86cb31ac-0843a890/Home/hassos_ova-3.5.vmdk" : failed to open (Object type requires hosted I/O).
2019-11-11T03:39:17.590Z| vmx| I125: DISKLIB-DSCPTR: Opened [0]: "hassos_ova-3.5-s001.vmdk" (0x10a)
2019-11-11T03:39:17.590Z| vmx| I125: DISKLIB-LINK  : Opened '/vmfs/volumes/86cb31ac-0843a890/Home/hassos_ova-3.5.vmdk' (0x10a): twoGbMaxExtentSparse, 41943040 sectors / 20 GB.
2019-11-11T03:39:17.590Z| vmx| I125: DISKLIB-LIB   : Opened "/vmfs/volumes/86cb31ac-0843a890/Home/hassos_ova-3.5.vmdk" (flags 0x10a, type twoGbMaxExtentSparse).
 
I found a work around:
Shutdown the VM and check the disk with vmkfstools
```
[root@esx01:/vmfs/volumes/yahda/Home] vmkfstools -x check hassos_ova-3.5.vmdk
Disk is error free
```

Then cloned it
```
[root@esx01:/vmfs/volumes/yahda/Home] vmkfstools -i hassos_ova-3.5.vmdk -d thin hassos_ova-3.5thin.vmdk
Destination disk format: VMFS thin-provisioned
Cloning disk 'hassos_ova-3.5.vmdk'...
Clone: 100% done.
```

Drop the original disk from the vm and add the hassos_ova-3.5thin.vmdk
Snapshots or cloning of the VM won't work until you do this, something in the build process is messing up the VMDK.

closed 2020-01-19 by stale[bot]

markhaines, 2020-03-16:
Thanks for this - I had exactly same issue and was unable to backup using Active Backup on Synology DSM, followed the steps above worked perfectly for me.

1h8fulkat, 2020-05-12:
Thank you - this fixed my issues with snaps as well!

zsbobbob, 2020-11-26:
Thank you for sharing. It helped lot!

AdamGalt, 2020-11-28:
Thank you. I had the same issue and this workaround has solved it.

JFK344, 2021-09-02:
This should be mentioned in the installation docs 🤔. Would have helped me a lot to know.

audioe, 2021-12-16:
Thanks for this.  Managed to stumble across it after searching for hours on vpshere sites trying to determine why I couldn't consolidate disks.

blastehh, 2024-01-26:
Just ran in to this issue... how is this not fixed?