home-assistant/operating-system #500: RPI4 falsely reporting SD Card Full

tungmeister, 2019-12-05:
I've just migrated to my Pi4 and there's an issue in that I can't create snapshots or any larger files as the OS reports that there's insufficient disk space. I've tried multiple SD cards but it seems that once Hassos has been rebooted a few times the partitions start behaving as though they're full. I have no issue copying over my snapshot for initial restore but after that no more can be created and If I delete that snapshot the disk still behaves as though it's full.

tungmeister, 2019-12-10:
Just to add to this, I decided to see what would happen if I kept home assistant running on the pi4. I've after attempting to update an add-on home assistant hung. I rebooted the pi and all my add-ons have disappeared from the side menu and I can't access the hass.io page anymore. As I said I've tried this with more than one sdcard and my setup works perfectly on the pi3b+ so it must be a pi4 issue

DeliriousMetro, 2019-12-13:
I recently moved from my RPI2 HassOS 2.12 to HassOS 3.5 on my RPI4 with HA 100.2,100.3,102.3 and now 103.0 and I have been able to backup three full snapshots and about a couple partial snapshots and use the Google Drive addon to automatically do so.

My only issue has been my database has been getting corrupted quite often - but it could just be my microSD card from 2015 wearing out. I will probably disable the recorder or just try using an SSD with a docker install of HA so that the wear cycles are not a issue.

My setup is working fine and Hass.io reports I have used 6.5% of my 64GB Sandisk extreme microsd and it hasn't been moving up nor down a lot using the systemmonitor 'disk_use_percent' sensor. I have updated addons, configured sensors and automations - updated HA as well multiple times.

The Rpi4 HassOS is RC release, but what you are describing is very strange indeed. I have rebooted numerous times and reconfigured so much that I do not get any disk full issues. I do have the 2GB variant if that might troubleshoot anything 🤷‍♂️

Is there anything in your logs that might diagnose something?

pvizeli, 2019-12-13:
Mabye an issue on filesystem expand. I modify that a bit on 3.6, could be that fix it

tungmeister, 2019-12-13:
I've been using the 32bit release on a 4gb RPi4. I've tried with both a 16gb and 200Gb SD card with the same results. I've moved back to the 3b for now. I'll migrate back over to the Pi4 this weekend and get some logs. @pvizeli is there a 3.6 image available?

Fuzzzy-HASSIO, 2019-12-31:
Are these issues still current?? I have HASSIO setup on a Pi 3b.. I just bought a 4 and am trying to decide if I should migrate my HA to the 4 or use the 4 to stream media..

closed 2020-03-08 by stale[bot]