home-assistant/operating-system #506: Audio device support

DaAwesomeP, 2019-12-11:
Hello,

I see in the NUC kernel config only a few audio drivers/codecs are being included. I am using a NUC-like device and so far everything (WiFi, Bluetooth, etc.) except audio is working. PCI device is `1002:4383` which is apparently an AMD SBx00 Azalia (Intel HDA) soundcard. There is also an HDMI soundcard device, but that that is less important.

I am seeing the following errors, and ALSA doesn't see any soundcards:
```
hdaudio hdaudioC0D0: Unable to bind the codec
hdaudio hdaudioC1D0: Unable to bind the codec
```

I probably need one of the following (hard to tell):
```
CONFIG_SND_HDA_GENERIC=y
CONFIG_SND_HDA_CODEC_REALTEK=m
CONFIG_SND_HDA_CODEC_ANALOG=m
CONFIG_SND_HDA_CODEC_CIRRUS=m
CONFIG_SND_HDA_CODEC_CONEXANT=m
```

I would like to test adding more codecs, but I cannot find in the docs how to build images other than the OVA image. What is the process for building the NUC image.

I know that this device is not a NUC, but this is the only x86 hardware (not virtualized) image offered, and supporting more devices would also allow for USB soundcards.

DaAwesomeP, 2020-01-08:
Not sure who to ping here, but I would appreciate some help!

pvizeli, 2020-01-09:
https://github.com/home-assistant/hassos/blob/dev/Documentation/getting_started_development.md

You can make the PR again the intel nuc kernel defconfig

DaAwesomeP, 2020-01-09:
Thanks for the link! I tried building from master (without any modifications so I could make sure I could build), and I got nearly to the end, and then I got this error (also tried without `j` flag, same result):
```
config intel_nuc
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "intel_nuc_defconfig"
make[1]: Entering directory '/build/buildroot'
#
# configuration written to /build/buildroot/.config
#
make[1]: Leaving directory '/build/buildroot'
build intel_nuc
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
make[2]: warning: -jN forced in submake: disabling jobserver mode.
make[2]: Leaving directory '/build/buildroot'
>>> hassos 1.0.0 Installing to target
docker run --rm --privileged -e BUILDER_UID="0" -e BUILDER_GID="0" -v /build/buildroot/output/images:/export hassos-hostapps --supervisor "homeassistant/amd64-hassio-supervisor" --supervisor-version "187" --supervisor-args "-e HOMEASSISTANT_REPOSITORY=homeassistant/intel-nuc-homeassistant" --supervisor-profile "hassio-supervisor" --supervisor-profile-url "http://s3.amazonaws.com/hassio-version/apparmor.txt" --cli "homeassistant/amd64-hassio-cli" --cli-version "14" --cli-args "--network=hassio --add-host hassio:172.30.32.2" --cli-profile "docker-default" --cli-profile-url "" --apparmor "supervisor/apparmor"
1+0 records in
1+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 47.3802 s, 22.7 MB/s
mke2fs 1.44.1 (24-Mar-2018)
Discarding device blocks: done
Creating filesystem with 262144 4k blocks and 65536 inodes
Filesystem UUID: 47cbc2a2-3de5-4a9e-8e65-59586635a3fc
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

mount: /mnt/data: failed to setup loop device for /export/data.ext4.
package/pkg-generic.mk:331: recipe for target '/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed' failed
make[1]: *** [/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed] Error 32
make[1]: Leaving directory '/build/buildroot'
Makefile:24: recipe for target 'intel_nuc' failed
make: *** [intel_nuc] Error 2
```
I restarted the host and did not run `make clean` and tried again, and now I get this:
```
config intel_nuc
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "intel_nuc_defconfig"
make[1]: Entering directory '/build/buildroot'
#
# configuration written to /build/buildroot/.config
#
make[1]: Leaving directory '/build/buildroot'
build intel_nuc
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
make[2]: Leaving directory '/build/buildroot'
>>> hassos 1.0.0 Installing to target
docker run --rm --privileged -e BUILDER_UID="0" -e BUILDER_GID="0" -v /build/buildroot/output/images:/export hassos-hostapps --supervisor "homeassistant/amd64-hassio-supervisor" --supervisor-version "187" --supervisor-args "-e HOMEASSISTANT_REPOSITORY=homeassistant/intel-nuc-homeassistant" --supervisor-profile "hassio-supervisor" --supervisor-profile-url "http://s3.amazonaws.com/hassio-version/apparmor.txt" --cli "homeassistant/amd64-hassio-cli" --cli-version "14" --cli-args "--network=hassio --add-host hassio:172.30.32.2" --cli-profile "docker-default" --cli-profile-url "" --apparmor "supervisor/apparmor"
Unable to find image 'hassos-hostapps:latest' locally
docker: Error response from daemon: pull access denied for hassos-hostapps, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
package/pkg-generic.mk:331: recipe for target '/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed' failed
make[1]: *** [/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed] Error 125
make[1]: Leaving directory '/build/buildroot'
Makefile:24: recipe for target 'intel_nuc' failed
make: *** [intel_nuc] Error 2
```
Any ideas? It seems like there is some Docker bug related to loops ([possibly this?](https://github.com/moby/moby/issues/27886)), but I'm sure this works for most people. My host is Debain Bullseye. I could try Ubuntu if that is known to work. Is the second error because it didn't finish writing to the target? Is there a way to clean only that step so I don't have to rebuilt everything again?

pvizeli, 2020-01-10:
You jump into container and enter: make intel_nuc

DaAwesomeP, 2020-01-10:
OK, I completely cleared docker images and containers and restarted the host and tried again overnight and it worked this time. Didn't change anything, but it worked. Now I'll try the audio kernel stuff.

DaAwesomeP, 2020-01-10:
I got it to compile and I now see what appears to be the analog in and out and the displayport out on my machine. However, I tried `spotify_connect`, and I can't get it to make any sound. What is the easiest way to get it to make a sound? Is there a way to access the ALSA or PortAudio mixer?
[image]

DaAwesomeP, 2020-01-10:
Spotify Connect is showing `SB: 0` as both the input and output, but should it be listing 3? At least one is the input, one is the output, and the last is something else. I think the generic one is the DisplayPort output because it isn't listed in the input select box.
[image]

alignus-abe, 2020-01-17:
I have a generic Intel NUC but no audio devices. I ensured the BIOS had audio output enabled.

Here's the HassOS config:
```
serial:
 /dev/ttyUSB0
 /dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F00A1E-if01-port0
 /dev/ttyUSB1
 /dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F00A1E-if00-port0
 /dev/ttyS2
 /dev/ttyS3
 /dev/ttyS0
 /dev/ttyS1
input:
 Sleep Button
 Power Button 
 Video Bus
disk:
gpio:
audio:
```

I [opened as  #issue](https://github.com/home-assistant/hassio-supervisor/issues/1436) but wonder if it's the same as this?

[image]

closed 2020-02-03 by pvizeli

DaAwesomeP, 2020-02-06:
@pvizeli this is not entirely resolved. I upgraded to 3.9, and I am still having the issue that I had with my custom build.

I think the issue may be that it is only listing the first SB device, which I think is an input. Is it possible that since it has assigned a `chan_id` of `0` for all of the connections that it is choosing the wrong one? Here is my audio info from the console:
```
hassio > hardware info
audio:
  "0":
    devices:
    - chan_id: "3"
      chan_type: digital audio playback
    - chan_id: "0"
      chan_type: hardware dependent
    name: HDA-Intel - HD-Audio Generic
    type: Generic
  "1":
    devices:
    - chan_id: "0"
      chan_type: digital audio playback
    - chan_id: "0"
      chan_type: digital audio capture
    - chan_id: "0"
      chan_type: hardware dependent
    name: HDA-Intel - HDA ATI SB
    type: SB
```

I have attached new images below.
[image]
[image]

DaAwesomeP, 2020-02-17:
@pvizeli would this be an issue with the kernel config or with the supervisor?

DaAwesomeP, 2020-03-17:
So I upgraded to 3.12, and USB audio definitely work, it's just the built-in audio. Are there any additional settings to try?

DaAwesomeP, 2020-03-17:
The output selection seems to be fixed (it lists only one output), but no sound is produced.

DaAwesomeP, 2020-03-17:
OK, I just figured it out! It was muted by default. I ran `pactl set-sink-mute 0 0`

PR #560 (2020-02-03): Add more SND audio codec modules
Fix: #506