home-assistant/operating-system #507: "disabled due to remaining attempts" on Hyper-V VM

onlize, 2019-12-11:
Hello,

My Hyper-V VM worked for a few months without any issues. Now, for some reason, I cannot boot it. It goes to the menu:
[image]

When I select Shell and type boot, I see the following:
[image]

it loads correctly if I type 
```
boot system0
```

My question is why is it doing this and how can I make it work aromatically as it worked before. I do not want to go shell and type that command.

Is the problem because I see
```
disabled due to remaining attempts = 0
```
How to fix it?

Thank you.

pvizeli, 2019-12-12:
After boot, there should be a job they mark it system as correct. Try `rauc status mark-good`. What is the output of `rauc info`?

onlize, 2019-12-12:
What do you mean after the boot. When I go to number four (shell)? When I try commands, I see this:
[image]

pvizeli, 2019-12-13:
yes, after boot into host system, not on bootloader

closed 2020-02-18 by stale[bot]