home-assistant/operating-system #513: RPI4: Program located in /sbin folder is not found by udev rules evaluation

ocalvo, 2019-12-16:
Repro:
1. Flash new RPI4 with hassos 3.6
2. Insert a GSM Modem USB stick
3. Boot

[Expected]
New serial devices to be found in /dev/*USB*
[Actual]
usb_modeswitch fails to switch the USB stick and add the new serial devices

[Workaround]
Create an specifc udev rule with a hardcoded path to usb_modeswitch:
```
ACTION=="add" \
, ATTRS{idVendor}=="12d1" \
, ATTRS{idProduct}=="14fe" \
, RUN+="/sbin/usb_modeswitch -X -v 12d1 -p 14fe"
```

If the /sbin/ folder is removed from the udev rule, then the rule fails to execute.

pvizeli, 2019-12-17:
Sure, if they use the exec system call, there is no PATH or shell.

However, feel free to make a PR they add this on overlay rootfs like hmip. I'm fine with it

pvizeli, 2019-12-18:
https://github.com/home-assistant/hassos/tree/dev/buildroot-external/rootfs-overlay/usr/lib/udev/rules.d

ocalvo, 2019-12-18:
I need to understand first why this is only required in RPI 4 and NOT in a VM.
Plus this code is specific to my usb dongle brand and model.
The USB switch data package is supposed to resolve this issue.

pvizeli, 2019-12-18:
The only difference is the kernel on both systems

ocalvo, 2019-12-18:
Other than hardware and kernels, any other difference?
I wonder if this is something specific about my dongle or if other dongle also have the same problem.

pvizeli, 2019-12-19:
On RPi we use the raspbian kernel and on OVA we use an upstream config

ocalvo, 2020-02-17:
@pvizeli is udev the same in all buildroot config for all platforms?

pvizeli, 2020-02-17:
Yes

closed 2020-04-25 by stale[bot]