home-assistant/operating-system #516: Installation requires an internet connection

micolous, 2019-12-18:
From https://github.com/home-assistant (emphasis added):

> Open source home automation that puts local control and **privacy first**

Please make a version of the installation media that **does not** require an internet connection to install.

Alternatively, allow downloading the Docker container files separately, and side-loading into HassOS with a USB drive (which would allow easy, air-gapped updates).

The current state of affairs seems to contradict the stated project goals of "privacy first", because I have no idea what HassOS just downloaded. 😞

pvizeli, 2019-12-18:
That is no issue. Hass.io work only with internet because they need setup itself. You can look at the source code to see what is going on.

closed 2019-12-18 by pvizeli

micolous, 2019-12-19:
I don't understand your response.  I understand components need to be downloaded. Let me download them and provide them to the VM manually.

Every 1 GB that this thing downloaded cost me $10.  HassOS currently doesn't tell you in advance how much it wants to download.  If it were given as a sideloadable download, then I'd be able to make a judgement call.

I have no idea if the source code provided matches the (unsigned) binary distribution. I'd rather just run this in an isolated environment, away from anything critical.