home-assistant/operating-system #517: Keyboard input in HassOS CLI is jibberish

micolous, 2019-12-18:
It's pretty hard to troubleshoot installation issues at the command line when it's reading the wrong keyboard layout, and so your input comes out as jibberish.

And extra hard when the CLI (before `login`) doesn't support arrow keys, tab completion, or other nice `libreadline` functionality. ;)

pvizeli, 2019-12-18:
Hi @micolous that is a nice idea. Follow is the repository with the code. Would be love to see an PR with your improvments 👍 

https://github.com/home-assistant/hassos-cli

closed 2019-12-18 by pvizeli

micolous, 2019-12-19:
There's two issues here, which I think are being conflated:

1. There is no keyboard layout selection. This would need be part of the init scripts. `loadkmap` lacks any binary layout file ([example](https://dev.alpinelinux.org/bkeymaps/)), and typing out full paths with the wrong layout is extremely difficult.

   You'd also have this issue with a Swiss keyboard when you try to use quotation marks or `gzip` at the shell. 😄 

2. Readline functionality. I agree that it is in scope for the other project.  I only mention it because it makes typing with the wrong layout even harder, but I don't feel strongly about this right now.