home-assistant/operating-system #518: RaspBerry Pi 4 with PoE HAT, fan not working (at all)

Vonkenboer, 2019-12-21:
**HassOS release with the issue:**

Hostname | hassio
System | HassOS 3.7
Deployment | production

arch | aarch64
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.5
timezone | Europe/Amsterdam
version | 0.103.0
virtualenv | false

**Supervisor logs:**
not applicable

**Journal logs:**
not applicable

**Kernel logs:**
not applicable

**Description of problem:**
Fan of original Raspberry Pi PoE HAT is never activated, resulting in hot RasPi 4 device. Prior HasOs 3.5 had the same issue.

Vonkenboer, 2019-12-21:
Also reported by Domoticon on [Home Assistant Forum](https://community.home-assistant.io/t/raspberry-pi-4-on-poe-hat-getting-hot-and-fan-of-poe-hat-is-never-running)

jm66, 2020-02-04:
Same issue here. Tried adding the following line to `/mnt/boot/config.txt` but the fan does not work at all.

```
dtoverlay=rpi-poe,poe_fan_temp0=50000,poe_fan_temp1=55000,poe_fan_temp0_hyst=5000,poe_fan_temp1_hyst=5000
```

jm66, 2020-02-07:
Hi @pvizeli when you have a moment would you please point me in the right direction? I’ve tried many approaches but nothing has worked. Thank you!

pvizeli, 2020-02-07:
You can jump into the development and look at the history of how I update the RPi kernel and Firmware to try a custom build with newer firmware/kernel. Maybe you need to update the EEPROM of your RPI to support that device. However, I can't promise anything, the RaspberrryPi 4 have a lot of bad decision.

Vonkenboer, 2020-03-22:
> You can jump into the development and look at the history of how I update the RPi kernel and Firmware to try a custom build with newer firmware/kernel. Maybe you need to update the EEPROM of your [...]

How can I update the EEPROM? Just boot with Raspbian Buster and update?

mvoort1337, 2020-04-07:
I have the same issue on a RPI4 with the official POE HAT.
The fan works fine when it's running on a clean installation of buster from raspberrypi.org
When running Home Assistant with HassOS 3.12 the fan never comes up.

Then I read you comment @pvizeli and thought this could be an issue with the RPI4, so I went back to my RPI3 B+.
Now I find that when I run the 32-bit version, the fan from the POE HAT works fine, but when I use the 64-bit version, the fan doesn't come up.

Could this be an issue with the 64-bit kernel rather than the RPI4 board?

I haven't tested the 32 bit version on the RPI4, and will do that tomorrow.

I am happy to have a look at source files, but can you give me a pointer in the right direction on there I should start looking, I've had a look in this repo but don't know where to find the HassOS kernel source files.

Many thanks for your attention to this.
Kind regards,
Mats

mvoort1337, 2020-04-08:
Hi @pvizeli 
I've ran tests with HassOS 3.12, clean image installation on a Raspberry Pi Model B+ and a Raspberry Pi 4 Model B.
Both RPIs have the official Raspberry Pi POE Hat
Both RPIs are powered over Ethernet from a Uniquiti Switch with 802.3af

RPI3 B+
32-bits HassOS: Fan works as expected.
64-bits HassOS: Fan never starts.

RPI4 B
32-bits HassOS: Fan works as expected.
64-bits HassOS: Fan never starts.

This may be an indication that the issue is with the 64-bits version of HassOS.
Can you let me know if there are other tests I can do to help confirm or rule this out?

G8YTZ, 2020-04-14:
Hi @mvoort1337 

What you say makes perfect sense, I am running 64 Bit HASS.IO on a 2gb RPI v4. Hopefully there will be a fix soon.

pvizeli, 2020-04-15:
Yes, 64bit OS is not supported with Raspberry. Use the 64bit image only if you don't need explicit RaspberrPi support.

I also strong recommend to use an Odroid N4 with are 4x faster as an RPi4 if you don't need explicit an RPi. You can also use the 4 year old Odroic C2 with also have more power as an RPi4 and is pretty cheap.

fjufju, 2020-04-29:
@pvizeli do you mean Odroid N2? There is no N4.

G8YTZ, 2020-04-29:
That Odroid looks very interesting indeed.

pvizeli, 2020-04-29:
Yeah, N2

DaftHonk, 2020-06-03:
A similar issue was encountered here: TheRemote/Ubuntu-Server-raspi4-unofficial#15 
The solution was setting `CONFIG_SENSORS_RPI_POE_FAN m` to the config before building the kernel. I'm not familiar enough with buildroot to know if a similar thing could be done here.

ckhyatt, 2020-06-03:
I am migrating from a Pi4 to the N2. Does the HA N2 image install on the SPI flash or is emmc/sd recommended? Thanks!

pvizeli, 2020-06-08:
Maybe that will be solved later with RPi kernel. Since we use just the original kernel from raspberry and firmware, that is nothing what we can fix or care about

closed 2020-06-08 by pvizeli

Alchiadus, 2020-08-08:
The PoE HAT fan now works on 64-bit since HassOS 4.12.

G8YTZ, 2020-08-08:
h

> The PoE HAT fan now works on 64-bit since HassOS 4.12.

Great news, thanks for confirming! The 64 Bit version is much quicker.