home-assistant/operating-system #519: Intel NUC install fails on NVMe

xbmcnut, 2019-12-23:
**Home Assistant release with the issue:**
1.101.3

**Operating environment (HassOS/Generic):**
HassOS, Intel NUC5i3RYK i3-5010U with [256GB NVMe M.2 M Keyed SSD

**Supervisor logs:**
N/A

**Description of problem:**
When trying to install the Intel 2.12 NUC image on an internal NVMe SSD using the dd method below, 
`-c /media/sda1/hassos_intel-nuc-2.12.img.gz | dd of=/dev/nvmen01 bs=4M conv=fsync`
the installation stalls at the following line:
`clocksource: switched to clocksource tsc`

Swapping the drive out for a M.2 **Sata** SSD and trying again allows the install to complete. Tried 6 or 7 times on the M.2 NVMe without success. I then tried installing Proxmox on to the [NVMe](https://www.amazon.com/dp/B07KGLN3HN/ref=cm_sw_r_tw_dp_U_x_GF8YDbVN19G30) and this installs without issue. Another user has used the Etcher method with a NVMe drive and the install fails too at the same line.

Not sure if this is the right location for the issue so feel free to move.

lurv, 2020-01-11:
Same issue here on Intel NUC 5i3RYH with Corsair Force MP510 SSD 240GB M.2 NVMe

AdamNaj, 2020-02-05:
Same problem for me ([Screenshot](https://photos.app.goo.gl/5ahe6x1E2oPypi487))
- Intel NUC7I5BNB i5-7260U 
- Intel SSDPEKNW512GB PCIe 4x, 8.0GT/s - NVMe

pvizeli, 2020-02-06:
Doublicate of #422

closed 2020-02-06 by pvizeli