home-assistant/operating-system #520: RPi 3B bluetooth causes WiFi issues

scop, 2019-12-27:
**HassOS release with the issue:**

3.7 32-bit. Also tested with 3.7 64-bit, 2.5 64-bit, 2.12 32-bit, and 3.6 32-bit.

**Supervisor logs:**

```
18-12-21 18:54:04 INFO (MainThread) [__main__] Initialize Hass.io setup
18-12-21 18:54:05 INFO (MainThread) [hassio.docker.network] Can't find Hass.io network, create new network
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Home Assistant configuration folder /data/homeassistant
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io SSL/TLS folder /data/ssl
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io Add-on data folder /data/addons/data
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io Add-on local repository folder /data/addons/local
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io Add-on git repositories folder /data/addons/git
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io temp folder /data/tmp
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io backup folder /data/backup
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io share folder /data/share
18-12-21 18:54:05 INFO (MainThread) [hassio.bootstrap] Create Hass.io DNS folder /data/dns
18-12-21 18:54:05 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to Supervisor homeassistant/armv7-hassio-supervisor with version 193
18-12-21 18:54:05 INFO (SyncWorker_0) [hassio.docker.supervisor] Connect Supervisor to Hass.io Network
18-12-21 18:54:06 INFO (MainThread) [__main__] Setup HassIO
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager
18-12-21 18:54:06 INFO (MainThread) [hassio.host.info] Update local host information
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
18-12-21 18:54:06 INFO (MainThread) [hassio.host.services] Update service information
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
18-12-21 18:54:06 INFO (MainThread) [hassio.host.network] Update local network DNS information
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
18-12-21 18:54:06 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
18-12-21 18:54:06 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service
18-12-21 18:54:06 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
18-12-21 18:54:06 INFO (MainThread) [hassio.dns] No CoreDNS plugin Docker image homeassistant/armv7-hassio-dns found.
18-12-21 18:54:06 INFO (MainThread) [hassio.dns] Setup CoreDNS plugin
18-12-21 18:54:06 INFO (MainThread) [hassio.updater] Fetch update data from https://version.home-assistant.io/stable.json
18-12-21 18:54:07 WARNING (MainThread) [hassio.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:None [Address not available]
18-12-21 18:54:07 WARNING (MainThread) [hassio.dns] Error on install CoreDNS plugin. Retry in 30sec
19-12-27 07:32:22 INFO (MainThread) [hassio.updater] Fetch update data from https://version.home-assistant.io/stable.json
19-12-27 07:32:22 INFO (SyncWorker_0) [hassio.docker.interface] Pull image homeassistant/armv7-hassio-dns tag 1.
19-12-27 07:32:42 INFO (MainThread) [hassio.dns] CoreDNS plugin now installed
19-12-27 07:32:42 INFO (MainThread) [hassio.dns] Update resolv from Supervisor
19-12-27 07:32:42 INFO (MainThread) [hassio.misc.forwarder] Start DNS port forwarding to 172.30.32.3
19-12-27 07:32:42 INFO (MainThread) [hassio.dns] Start CoreDNS plugin
19-12-27 07:32:51 INFO (SyncWorker_2) [hassio.docker.dns] Start DNS homeassistant/armv7-hassio-dns with version 1
19-12-27 07:32:51 INFO (MainThread) [hassio.homeassistant] No Home Assistant Docker image homeassistant/raspberrypi3-homeassistant found.
19-12-27 07:32:51 INFO (MainThread) [hassio.homeassistant] Setup HomeAssistant landingpage
19-12-27 07:32:51 INFO (SyncWorker_1) [hassio.docker.interface] Pull image homeassistant/raspberrypi3-homeassistant tag landingpage.
19-12-27 07:33:06 INFO (MainThread) [hassio.hassos] Detect HassOS 3.7 on host system
19-12-27 07:33:06 INFO (SyncWorker_5) [hassio.docker.hassos_cli] Found HassOS CLI homeassistant/armv7-hassio-cli with version 15
19-12-27 07:33:06 INFO (MainThread) [hassio.store.git] Clone add-on https://github.com/hassio-addons/repository repository
19-12-27 07:33:06 INFO (MainThread) [hassio.store.git] Clone add-on https://github.com/home-assistant/hassio-addons repository
19-12-27 07:33:10 INFO (MainThread) [hassio.store] Load add-ons from store: 62 all - 62 new - 0 remove
19-12-27 07:33:10 INFO (MainThread) [hassio.addons] Found 0 installed add-ons
19-12-27 07:33:10 INFO (MainThread) [hassio.snapshots] Found 0 snapshot files
19-12-27 07:33:10 INFO (MainThread) [hassio.discovery] Load 0 messages
19-12-27 07:33:10 INFO (MainThread) [hassio.ingress] Load 0 ingress session
19-12-27 07:33:10 INFO (MainThread) [hassio.secrets] Load Home Assistant secrets: 0
19-12-27 07:33:10 INFO (MainThread) [__main__] Run Hass.io
19-12-27 07:33:10 INFO (MainThread) [hassio.api] Start API on 172.30.32.2
19-12-27 07:33:10 INFO (MainThread) [hassio.addons] Phase 'initialize' start 0 add-ons
19-12-27 07:33:10 INFO (MainThread) [hassio.addons] Phase 'system' start 0 add-ons
19-12-27 07:33:10 INFO (MainThread) [hassio.addons] Phase 'services' start 0 add-ons
19-12-27 07:33:18 INFO (SyncWorker_9) [hassio.docker.homeassistant] Start homeassistant homeassistant/raspberrypi3-homeassistant with version landingpage
19-12-27 07:33:23 INFO (MainThread) [hassio.homeassistant] Detect a running Home Assistant instance
19-12-27 07:33:23 INFO (MainThread) [hassio.addons] Phase 'application' start 0 add-ons
19-12-27 07:33:24 INFO (MainThread) [hassio.tasks] All core tasks are scheduled
19-12-27 07:33:24 INFO (MainThread) [hassio.core] Hass.io is up and running
19-12-27 07:33:24 INFO (MainThread) [hassio.homeassistant] Setup Home Assistant
[... 511 lines cut ...]
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1821 ttl=64 time=2314 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1822 ttl=64 time=2848 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1823 ttl=64 time=3376 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1824 ttl=64 time=3117 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1825 ttl=64 time=2119 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1826 ttl=64 time=1120 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1827 ttl=64 time=121 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1828 ttl=64 time=4.27 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1829 ttl=64 time=1360 ms
64 bytes from 192.168.100.254 (192.168.100.254): icmp_seq=1830 ttl=64 time=333 ms
```
...and sometimes ping loss, and even higher times. Naturally, net access of HA including SSH and web is very shaky in this state.

I have tested this with three different RPi 3B's, and happens the same with all of them. Booting off USB SSD vs SD (and no other USB devices plugged) makes nSSH and web accesso difference. All three RPis have been USB boot enabled, i.e. `program_usb_boot_mode=1`.

I have successfully used bluetooth on some of these RPis before. But it is possible that at that time they have been connected via ethernet, not wifi. Can't remember. And it could have been even pre-HassOS (i.e. ResinOS) times.

I've found a couple of things that I could try out, but I'm not sure how do I go about it in a RPi3 HassOS setup, i.e. where exactly on the SD card to add the files:
https://github.com/raspberrypi/linux/issues/1552#issuecomment-457831706
https://github.com/greg-erskine/pCP-github/commit/0083f17e9725db03548351f713674145eeb83d8b
I suppose the modprobe option would be doable via cmdline.txt, will try that out.

Will also try with ethernet soon and report back.

Anyway, help and debugging instructions including where to try out the above settings would be welcome.

scop, 2019-12-27:
Also: nothing in home-assistant.log

scop, 2019-12-27:
`brcmfmac.feature_disable=0x96` in cmdline.txt makes no difference.

scop, 2019-12-27:
No problems with bluetooth+ethernet, so it's definitely the bluetooth+wifi combo that has issues.

smoke007, 2020-01-27:
Just throwing in my 2 cents on personal experience with the same issue.  I agree that I've also had this same wifi problem with RP3 and RP zero w when bluetooth_tracker is enabled.  Turning off bluetooth_track resolves the network connectivity problem.  I've found bluetooth_tracker is best used with a wired network connection, so you don't have network connectivity issues.

I also experienced the same wifi issues trying to do bluetooth tracking with this other project called Monitor on a RP zero w and RP3.  https://github.com/andrewjfreyer/monitor  I think it's either a hardware related issue with trying to use BT and wifi at the same time or both projects are using the same wifi and BT driver with the same bug.

scop, 2020-01-30:
I've moved on to Ethernet with the setup I saw this with, but I plan to try this out with HassOS 3.8 soon with another.

scop, 2020-04-04:
Still going to have a look sometime.

pvizeli, 2020-05-08:
You see the issue that I all time said, RaspberryPi is bad hardware. There are much better chip manufacturers around as Broadcom. On data paper, the RPi looks fine but most people don't know the price which this data papers have. You can never use al resources at the same time.

closed 2020-05-08 by pvizeli

scop, 2020-05-09:
I'm still not using bluetooth+wifi, but bluetooth was broken even with ethernet for a long time. It works again with 3.13, so now's again a time when it would make sense to test this again if someone feels like it. I don't have plans to move to wifi any time soon though.