home-assistant/operating-system #524: Raspberry Pi 3B "Error looking up Bluetooth device"

MortenNDKK, 2020-01-05:
**HassOS release with the issue:**

HassOS 3.7
Home Assistant (Hassio) 0.103.5
Supervisor 193

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I actived bluetooth as a device tracker, but it doesn't work. I get an error in my log stating:

Error looking up Bluetooth device
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 31, in discover_devices
    lookup_class=lookup_class, device_id=device_id)
_bluetooth.error: (110, 'Operation timed out')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 140, in perform_bluetooth_update
    devices = await hass.async_add_executor_job(discover_devices, device_id)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 62, in discover_devices
    device_id=device_id,
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 34, in discover_devices
    raise BluetoothError ("error communicating with local "
bluetooth.btcommon.BluetoothError: error communicating with local bluetooth adapter

------
I can see on Home Assistant forum that we are at least two with the same problem.

talondnb, 2020-01-06:
https://github.com/home-assistant/home-assistant/issues/30147

talondnb, 2020-01-10:
Issue still apparent on 0.103.6.

talondnb, 2020-01-15:
Issue still apparent with HassOS 3.8 and supervisor 195.

Cooad, 2020-02-11:
Still doesn't work on HassOS 3.10.
There's a lot of good information in the issue mentioned 3 comments above.
The latest version of HassOS where Raspberry's build-in bluetooth works correctly is 2.12.
Someone tracked it down to a rasberry pi firmaware bug - maybe a firmware update in the next HassOS release would fix it? :)

r-jordan, 2020-02-14:
I found a workaround to this problem here:

https://github.com/raspberrypi/firmware/issues/1150#issuecomment-581890960

iromeo, 2020-03-23:
@r-jordan 

This work around doesn't work in my case, same issue in ~ day.

```
ha > login
# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:**:**:**:**  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING
	RX bytes:21588252 acl:25 sco:188 events:639500 errors:0
	TX bytes:421476 acl:0 sco:0 commands:35854 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: SLAVE ACCEPT
Can't read local name on hci0: Connection timed out (110)
# ps aux | grep hciat
root     11581  0.0  0.0   1812     0 ?        S    Mar21   0:00 hciattach /dev/serial1 bcm43xx 38400 noflow - b8:27:eb:c7:34:73
root     12225  0.0  0.0   3180   476 pts/0    S+   08:02   0:00 grep hciat

# uname -a
Linux ha 4.19.106-v7 #1 SMP Fri Mar 6 19:35:05 UTC 2020 armv7l HassOS
```

lucagiove, 2020-03-24:
Hi @pvizeli as described here:
https://github.com/raspberrypi/firmware/issues/1150
and
https://github.com/home-assistant/core/issues/30147
I'd like to submit a Pull Request with a modified version of the btuart file.
How would you suggest to prepare a patch? I'm not used to buildroot but I'm familiar with other build systems like Yocto.
Thanks, Luca

lucagiove, 2020-04-03:
I did those PR: #614 #616 
I'm currently testing it and seems to be working if you wanna try a test image for rpi3b with my fix below the link to the image:
https://drive.google.com/open?id=1ue0tELFgTZys-_KRs9CSZoq9bhUx9OYF

It's an unofficial version built from development branch.

closed 2020-06-08 by pvizeli

pilot1981, 2020-10-21:
in 0.116.4 there is again the problem...

lorisbc, 2020-10-21:
You can revert to 4.13 with `ha os update --version 4.13`
This issue will be fixed in 4.15 -> home-assistant/operating-system#910