home-assistant/operating-system #525: Import from USB on RPi4 doesn't set i2c parameters inside /boot/config.txt

crag-r, 2020-01-05:
**HassOS release with the issue:**
HassOS 3.7

**Supervisor logs:**
`Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'`

**Host Logs:**
`ls -l /dev/i2c*:
ls: /dev/i2c*: No such file or directory`

**Description of problem:**
Following instructions [Here](https://www.home-assistant.io/hassio/enable_i2c/) to enable i2c works fine for RPi 3, using the same USB stick on the RPi 4, it doesn't update /boot/config.txt therefore doesn't enable i2c
Manually uncommenting dtparam=i2c_arm=on and adding dtparam=i2c1=on then rebooting the host enables i2c. 
`ls -l /dev/i2c*
crw-------    1 root     root       89,   1 Jan  5 19:43 /dev/i2c-1`

I have just tested on a completely fresh Hassos 3.7 install, plugged the usb stick in and checked /etc/modules-load.d/ nothing there as expected, clicked import from USB inside Hass.io and interestingly rpi-i2c.conf **does** get placed in /etc/modules-load.d/ so the usb import is sort of working, it just doesn't modify the config.txt inside /boot/config.txt

Has anyone any ideas? is there any logs I can provide from journal on the host that will help?

Thanks

pvizeli, 2020-01-06:
`just doesn't modify the config.txt inside /boot/config.txt`

Yes, they import don't touch the `config.txt`. So it's not directly a bug as a feature request.

crag-r, 2020-01-09:
I have just checked what the import has access to and yes you are correct it doesn't touch the config.txt [link](https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md#automatic)

Looks like the instructions on home-assistant.io need changing on how to enable i2c, unless the import used to touch the config.txt in an older hassOS release?

pepeEL, 2020-01-09:
Or maybe better will be add this function to new image hassos ? To import config can mod /boot/config.txt

pepeEL, 2020-01-09:
also setting for NTP is wrong path:

NTP
You can manual edit the systemd timesync file on /etc/systemd/timesyncd.conf. Our default ntp configuration look like:

[Time]
NTP=time1.google.com time2.google.com time3.google.com
FallbackNTP=0.pool.ntp.org 1.pool.ntp.org 2.pool.ntp.org 3.pool.ntp.org

In Hassos not found this path: /etc/systemd/timesyncd.conf.

closed 2020-03-16 by stale[bot]