home-assistant/operating-system #526: [RPi4 4GB] hassOS 3.7 USB devices disconnected after some hours of uptime

JANogueira, 2020-01-07:
**HassOS release with the issue:**

Home Assistant 0.103.6

Version | 193
Latest version | 193

Hostname | hassio
System | HassOS 3.7
Deployment | production

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**
```
[  303.329705] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.                                                                                                                                                 
[  303.345777] xhci_hcd 0000:01:00.0: Host halt failed, -110                                                                                                                                                                             
[  303.345791] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead                                                                                                                                                   
[  303.345841] xhci_hcd 0000:01:00.0: HC died; cleaning up                                                                                                                                                                               
[  303.345984] usb 1-1: USB disconnect, device number 2                                                                                                                                                                                  
[  303.346009] usb 1-1.3: USB disconnect, device number 3
```

**Description of problem:**
After some hours, the USB devices get disconnected.

List of USB devices when the sistem boots up:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

After a couple of hours:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

It is only possible to recover visibility and control of them after system restart.
What other info/logs do you need?

Thanks in advance.

JANogueira, 2020-01-07:
more dmesg output:
```
➜  ~ dmesg | grep -i usb
[    0.197545] usbcore: registered new interface driver usbfs
[    0.197606] usbcore: registered new interface driver hub
[    0.197718] usbcore: registered new device driver usb
[    0.487955] usbcore: registered new interface driver r8152
[    0.488025] usbcore: registered new interface driver lan78xx
[    0.488474] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    0.495448] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    0.495481] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.495507] usb usb1: Product: xHCI Host Controller
[    0.495528] usb usb1: Manufacturer: Linux 4.19.88-v8 xhci-hcd
[    0.495549] usb usb1: SerialNumber: 0000:01:00.0
[    0.496073] hub 1-0:1.0: USB hub found
[    0.496671] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    0.496706] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    0.497145] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 4.19
[    0.497175] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.497200] usb usb2: Product: xHCI Host Controller
[    0.497220] usb usb2: Manufacturer: Linux 4.19.88-v8 xhci-hcd
[    0.497242] usb usb2: SerialNumber: 0000:01:00.0
[    0.497711] hub 2-0:1.0: USB hub found
[    0.499380] usbcore: registered new interface driver uas
[    0.499478] usbcore: registered new interface driver usb-storage
[    0.499601] usbcore: registered new interface driver usbserial_generic
[    0.499647] usbserial: USB Serial support registered for generic
[    0.507353] usbcore: registered new interface driver usbhid
[    0.507361] usbhid: USB HID core driver
[    0.830300] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    0.980982] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.20
[    0.981020] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    0.981041] usb 1-1: Product: USB2.0 Hub
[    0.982796] hub 1-1:1.0: USB hub found
[    1.278306] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[    1.615672] usbcore: registered new interface driver brcmfmac
[    1.999992] usb 1-1.3: New USB device found, idVendor=0463, idProduct=ffff, bcdDevice= 0.01
[    2.000016] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.000027] usb 1-1.3: Product: 5E
[    2.000036] usb 1-1.3: Manufacturer: EATON
[    3.967565] hid-generic 0003:0463:FFFF.0001: hiddev96,hidraw0: USB HID v1.10 Device [EATON 5E] on usb-0000:01:00.0-1.3/input0
[  172.256228] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[  172.272306] xhci_hcd 0000:01:00.0: Host halt failed, -110
[  172.272321] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[  172.272369] xhci_hcd 0000:01:00.0: HC died; cleaning up [  172.272442] usb 1-1: USB disconnect, device number 2
[  172.272467] usb 1-1.3: USB disconnect, device number 3
```

JANogueira, 2020-01-07:
This drop is seen when frequent requests to the USB bus are made. On my situation, this is about NUT server poll frequency of 5 seconds that leads to xHCI problem, leading to USB connectivity loss.

If NUT server is disabled, or the poll frequency is larger than 5 seconds, this problem is not observed.
Is it normal to have a crash when configuring frequent USB poll?

JANogueira, 2020-01-08:
Another update:
After some observation, this seems not related with the requests made to the USB bus.
I changed the USB cable by a brand-new one, and also added a USB hub in between, but the problem is still visible.

This can happen right after host boot or after some hours. Either way, seems that the USB driver is the cause.

pvizeli, 2020-01-08:
I can update the kernel (we us the same as raspbian) maybe that fix it

JANogueira, 2020-01-08:
If you can do that, yes please.
Thanks in advance.

pvizeli, 2020-01-08:
I tested an RPi4 and can't reproduce your issue. Please a bit more details. If I can't reproduce it, I can't also fix it.

JANogueira, 2020-01-08:
What log levels can I enable in order to provide you all the details you need?

pvizeli, 2020-01-13:
The 3.8 have the latest RPi kernel and firmware include

JANogueira, 2020-01-13:
Many thanks.
I will test it and share the feedback.

JANogueira, 2020-01-13:
unfortunately, it is still happening exactly in the same way as seen before.
```
[  600.808456] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.                                                                                                                                                 
[  600.824533] xhci_hcd 0000:01:00.0: Host halt failed, -110                                                                                                                                                                             
[  600.824556] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead                                                                                                                                                   
[  600.824627] xhci_hcd 0000:01:00.0: HC died; cleaning up                                                                                                                                                                               
[  600.824853] usb 1-1: USB disconnect, device number 2                                                                                                                                                                                  
[  600.824888] usb 1-1.4: USB disconnect, device number 6                                           
```

pvizeli, 2020-01-13:
I would suggest opening an issue on https://github.com/raspberrypi/linux with all information from archlinux and this issue.

Since we use this downstream Linux kernel, they maintain the kernel.

JANogueira, 2020-01-13:
ok Thanks.

closed 2020-01-13 by JANogueira

JANogueira, 2020-01-13:
Hi @pvizeli ,

Are you including any EEPROM images or bootloader from Raspbian into HassOS?
I am asking this because i've just flashed the latest available EEPROM image (vl805 0137ad) that is aimed to fix USB instability, and all was good in Raspbian, but not in hassio (still observing xHCI crash).

Thanks in advance for your clarification.

JANogueira, 2020-01-13:
Please, disregard my last post.
I was doing it wrong while trying to flash the EEPROM. All good apparently. 3 hours have passed and all stable.

JANogueira, 2020-01-15:
Just to confirm that everything is still working perfectly.
rpi4b 0137ad eeprom solved the issue. USB devices are all stable and I had a small increase on the sdcard I/O.