home-assistant/operating-system #527: raspberry pi hassos image - how I can extand partition tu use full sdcard space?

lpt2007, 2020-01-08:
Raspberry pi hassos image - how I can extand partition tu use full sdcard space?

pvizeli, 2020-01-08:
It's done automatically

pepeEL, 2020-01-08:
Hmmm i think that no automatically because if i good remember i remove card from RPI and insert it to PC and partition is not full resize....

pvizeli, 2020-01-08:
I run a test with 3.7 and it work perfect. Also with OVA image. However, your PC need support hyprid partition, otherwise you see the wrong partition layout.

closed 2020-01-08 by pvizeli

lpt2007, 2020-01-08:
I try with fresh install of hassos_rpi2-3.7.img on 8Gb sdcard:

I can use only 6.6Gb:

[image]