home-assistant/operating-system #529: HassOS on Proxmox - not mapped USB ports...

darekhome, 2020-01-09:
**Home Assistant release with the issue:**
 Home Assistant 0.103.5

**Operating environment (HassOS/Generic):**
 System HassOS 3.7

**Supervisor logs:**

**Description of problem:**

Newest 'hass.io' on newest 'proxmox'.
Hardware 'Intel-NUC'.
Device connected to USB port, but not seen in 'hass.io'.
The same tested on 'plain PC' (DELL OptiPlex 780) - problem the same.
Only seen 4 'ttySx' ports (x = 0...3).

[image]

On 'proxmox' the device is enabled and seen:

[image]

darekhome, 2020-01-09:
It is Proxmox 6.1 with kernel 5.3...
And I use the highest version of the kernel with hass.io...

pvizeli, 2020-01-10:
I means what kernel option/module are needed for it?

darekhome, 2020-01-10:
With hass.io installed on ex. IntelNUC no any special options are required. The device is seen something like ttyUSB0.
But when (has.io) installed with proxmox nothing is seen...

So simply answering your question - no special kernel option is needed.

szurr, 2020-01-10:
Hi 
In my case workflow looks
1.Proxmox add devices (USB port/vendor)
2.Proxmox restart ! Whole vm/lxc
3. Hassio I see USB device and make changes to config
4. Restart and working (most of the time)

darekhome, 2020-01-11:
Well, this is what I expected...
But in my case it is not working.
I hass.io I always see (ONLY) ttyS0,...,ttyS3. Under hardware button.

pvizeli, 2020-01-11:
Look like our kernel need a module for this. The USB redirection on vmware/hyperV work. The question is only, which module would be needed on the kernel side...

pvizeli, 2020-01-11:
I put this to next release, maybe it work. I can't test it

pvizeli, 2020-01-11:
If someone will test it `hassio os update --version 3.8`

Iam-lcl, 2020-01-12:
Tested with a clean install of 3.8
no USB showed up.

pvizeli, 2020-01-13:
We test it with our available hardware and it worked. Can you ask the manufacturer of your device, which Linux kernel option they need on his hardware? So we can add it to our device kernel configs and make it work on all images.

darekhome, 2020-01-13:
The only info available on Internet is here: https://lore.kernel.org/patchwork/patch/625115/
https://lore.kernel.org/patchwork/patch/632848/

P.S. I have contacted the local 'honeywell.com' about device HGIU-80. But they said (according documentation they have) that there is no any requirements/limits to use this device. It should work with 'any' software... Whatever that means. But I'm not sure if they understood what I'm taking about...

pvizeli, 2020-01-13:
@darekhome thanks for the link. I created a PR they add the support for this chip to all HassOS platforms. They is included to 3.9

closed 2020-01-13 by pvizeli

darekhome, 2020-01-13:
Sound gorgeous ! Thank you for magnificent job !!!

Iam-lcl, 2020-01-13:
Nice!
But will this make my nuc with proxmox 6.1 be able to share the usb-ports with the guest hassOS?

pvizeli, 2020-01-14:
You should use the ova and not the nuc image @lam-lcl. The nuc image have no agent or support any virtualization optimize like USB forward.

However, if someone will test it with an ova image! -> `hassio os update --version 4.0`

darekhome, 2020-01-14:
O.K. Checked on other machine - looks PERFECT!!!
USB device appeared...
The only concern I have - it is appearing 'twice'. One as '/dev/ttyUSB0 and 2nd as '/dev/serial/by-id/...'.
Is that O.K.?

[image]

All above with USB 'pass-through' in proxmox.
[image]

Iam-lcl, 2020-01-14:
Ok,
Tested it out with my NUC (this one is a older first gen, with i3)
Hass.io is installed with whiskerz007 install script: https://github.com/whiskerz007/proxmox_hassos_install

This installs the latest release and it is using the ova. After `hassio os update --version 4.0`
`hassio os info` gives:
board: ova
boot: B
version: "4.0"
version_cli: "15"
version_cli_latest: "15"
version_latest: "3.8"

`hassio hardware info` gives:
audio: {}
disk: []
gpio: []
input:
- Power Button
- ImExPS/2 Generic Explorer Mouse
- AT Translated Set 2 keyboard
- QEMU QEMU USB Tablet
serial:
- /dev/ttyS0
- /dev/ttyS3
- /dev/ttyS2
- /dev/ttyS1

In proxmox I added the USB both as device and the whole port, (this works with debian VM).
The usb is at the moment a usb-flash drive.

What can I have missed?

darekhome, 2020-01-15:
Are you also talking about different device than HGI-80? Some USB-flash drive? Right?
The whole issue reported here was related to have support for 'TI USB 3410 or 5052 serial devices'.
As this cheap is used in the Honeywell HGI-80.

Your issue can be something completely different...

Iam-lcl, 2020-01-15:
Well yes, I believe so, as this issue started with 

> HassOS on Proxmox - not mapped USB ports

 a more similar issue to mine but later got changed to TI 3410/5052, so probably better to start a new issue.
My case is that no USB is mapped at all to HassOS. flash drives, TI CC2531 (zigbee usb) etc

phsluis, 2020-02-11:
Not sure if this is the right place for my post but I am using HassOS 3.10 and when I connect the HGI80 to the RPI 4 I get the error message that the TI 3410/ 5052 firmware cannot be loaded.

Did I do something wrong?

darekhome, 2020-02-12:
The error message is from 'Proxmox' or from 'HassOS'?

PR #534 (2020-01-11): VirtIO MMIO kernel module
Maybe fix: #529
PR #542 (2020-01-13): Add support for TI USB 3410 or 5052 serial devices
Fix: #529