home-assistant/operating-system #538: HassOS on RPi4 does not boot

jaakla, 2020-01-13:
HassOS on RPi4 does not boot. Console responds when I connect/disconnect devices (like USB keyboard), but does not proceed until login. Stays forever (at least 8 hours) to attached screenshot. Any hope to recover before full reinstall option?

Steps before it happened:
 1. misconfigured homeassistant to allow user-based access and got no access in any level. Perhaps the issue I put to trusted_users = username where I used visual username not ID as instructed.
 2. powered off RPi down, took SDCard with OS and modified ha main configuration directly on data partition, to remove the auth_providers section
 3. powered RPi up again, did not go to network
 4. checked direct console (connected with HDMI monitor), got attached image
 5. Did repair to the SD card file system, same situation.

[image]

pvizeli, 2020-01-13:
Hard to say. `Did repair to the SD card file system, same situation.` because of the hybrid partitions is that very critical to do. If your OS did not support this correctly, everything looks good but at same time everything breaks.

If you get no console access over the USB key, the docker is not correctly running and without docker all is dead. Maybe an SD card issue or corrupt docker database because of power down.

Maybe first the issue is hidden now and with all steps, it shows now a completely different issue.
Sorry for your issues. I would suggest now to download the latest 3.8 and reflash the SD card.

jaakla, 2020-01-13:
Thanks for quick response. Btw I used Paragon on Mac for the ExtFS mount, edit and also later repair; maybe that broke something. Some were suggesting this method with similar issue and it helped them. Boot issue came after first edit already, before FS Repair. I did not try console with boot before even editing, maybe the issue was there even then.

Is there a way to rescue individual containers, or at at least their user files ? e.g. where to look for my Node-red scripts and Grafana dashboard sources?

pvizeli, 2020-01-13:
Yes, that will be the issue. Please only edit files on boot partition. Also mac did not support hyprid partition layouts.

closed 2020-01-13 by pvizeli