home-assistant/operating-system #543: Allow storage=volatile for journald

jdeluyck, 2020-01-15:
**HassOS release with the issue:**
3.8

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
Currently HassOS is configuring journald persistent storage (storage=auto and /var/log/journal exists). This causes additional wear & tear on the SD card, and it would be great if people that know what they're doing could easily disable this (changing storage to volatile for only in-memory storage)

pvizeli, 2020-01-23:
If you want do this:
- Look if you can change the config with dbus
- Add support for this dbus on supervisor
- Make it configurable on supervisor

jdeluyck, 2020-01-23:
> If you want do this:
> 
> * Look if you can change the config with dbus
> * Add support for this dbus on supervisor
> * Make it configurable on supervisor

Might have a look at what's possible.

pvizeli, 2020-03-24:
I'm also fine to move that to data partition...

pvizeli, 2020-05-08:
Fix with 4.6

closed 2020-05-08 by pvizeli