home-assistant/operating-system #544: home assistant update errors

lpt2007, 2020-01-15:
I first update hassos to 3.8

20-01-15 11:28:00 INFO (SyncWorker_6) [hassio.docker.interface] Pull image homeassistant/qemux86-64-homeassistant tag 0.103.6.
20-01-15 11:28:16 ERROR (SyncWorker_6) [hassio.docker.interface] Can't install homeassistant/qemux86-64-homeassistant:0.103.6 -> 404 Client Error: Not Found ("no such image: homeassistant/qemux86-64-homeassistant:0.103.6: No such image: homeassistant/qemux86-64-homeassistant:0.103.6").
20-01-15 11:28:16 WARNING (MainThread) [hassio.homeassistant] Update Home Assistant image fails

ondoteam, 2020-01-16:
Yeah. Same error message from 3.7 to 3.8  using Intel NUC.

robbinjanssen, 2020-01-16:
Same on pi4 from 3.7 to 3.8, after this the error changed to:
`WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_rpi4-3.8.raucb: `

pvizeli, 2020-01-16:
> Yeah. Same error message from 3.7 to 3.8 using Intel NUC.

Can you restart the host and try it again?

ondoteam, 2020-01-16:
> > Yeah. Same error message from 3.7 to 3.8 using Intel NUC.
> 
> Can you restart the host and try it again?

The first law of computer science 🥇 . I have restarted and then updated without problem using a Intel NUC. Is this the normal procedure? Thank you @pvizeli

MichaelSchemmer, 2020-01-16:
Tried that, still not working.....

robbinjanssen, 2020-01-17:
I tried rebooting the host (`hassio host reboot`) and retried afterwards, still the same error;
```
20-01-17 08:48:13 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service
20-01-17 08:48:58 INFO (MainThread) [hassio.hassos] Detect HassOS 3.7 / BootSlot A
20-01-17 08:48:58 INFO (SyncWorker_4) [hassio.docker.hassos_cli] Found HassOS CLI homeassistant/armv7-hassio-cli with version 15
20-01-17 08:49:00 INFO (MainThread) [hassio.hassos] Rauc: A - marked slot kernel.0 as good
20-01-17 08:50:33 INFO (MainThread) [hassio.api.security] /hassos/info access from core_ssh
20-01-17 08:50:41 INFO (MainThread) [hassio.api.security] /hassos/update access from core_ssh
20-01-17 08:50:41 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_rpi4-3.8.raucb
20-01-17 08:55:41 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_rpi4-3.8.raucb: 
```

pvizeli, 2020-01-17:
@ondoteam I think there was an issue with resources/file descriptors. I linux reboot help there.

@robbinjanssen Can you download it manual with the link? The device can't download the file from this link.

MichaelSchemmer, 2020-01-17:
I'm running HASS on an intel NUC using Hyper-v and I receive the following error:
ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32

hadrianl, 2020-01-17:
same error

lpt2007, 2020-01-17:
First make full snapshot under hass.io and download it to your computer then go to hassos releases and download new image and flash it to sd card and start new system. Copy snapshot you made before to backup folder then go under snapshots and do a refresh. Then restore with wipe & restore. Wait to complete. Thats it:)

robbinjanssen, 2020-01-17:
> @ondoteam I think there was an issue with resources/file descriptors. I linux reboot help there.
> 
> @robbinjanssen Can you download it manual with the link? The device can't download the file from this link.

Yes I can;

```
$ ssh root@192.168.0.5

  _    _                 _       
 | |  | |               (_)      
 | |__| | __ _ ___ ___   _  ___  
 |  __  |/ _` / __/ __| | |/ _ \ 
 | |  | | (_| \__ \__ \_| | (_) |
 |_|  |_|\__,_|___/___(_)_|\___/ 
                                 

Our Cli:
$ hassio help

~ $ wget https://github.com/home-assistant/hassos/releases/download/3.8/hassos_rpi4-3.8.raucb
Connecting to github.com (140.82.118.4:443)
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (52.216.24.196:443)
saving to 'hassos_rpi4-3.8.raucb'
hassos_rpi4-3.8.rauc   2% |***                                                                                                                                                                 | 2015k  0:06:56 ETA
```

jensihnow, 2020-01-18:
Same (or similar) issue here while running on VMWare and HassOS 2.12:

```
20-01-18 15:47:37 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_ova-3.8.raucb
20-01-18 15:47:52 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-3.8.raucb
20-01-18 15:47:52 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-01-18 15:47:52 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-01-18 15:47:53 INFO (MainThread) [hassio.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-01-18 15:47:53 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-01-18 15:47:53 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32
```

Any additional data I can provide?

u17194907425, 2020-01-31:
```
20-01-31 19:00:37 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request running
20-01-31 19:00:44 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_ova-3.8.raucb
20-01-31 19:00:49 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_ova-3.8.raucb: Cannot connect to host github.com:443 ssl:None [Try again]
20-01-31 19:01:59 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_ova-3.8.raucb
20-01-31 19:02:04 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_ova-3.8.raucb: Cannot connect to host github.com:443 ssl:None [Try again]
```

However I can download that file with `wget` from within ssh connection to my device. Multiple reboots didn't do the trick. Any advice?

MichaelSchemmer, 2020-02-02:
Also working here after trying the update again, strange, didn't change anything

pvizeli, 2020-02-03:
> HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32

Look like a kernel issue (REL-2 use 4.14) with VM. Maybe it's fix with REL-3 (Linux 4.19).

rjm12rjm12, 2020-02-06:
I have the exact same setup and issue as jjmuriel.

jbct, 2020-02-06:
Seems there are multiple issues in this ticket. @robbinjanssen -  I have the same issue as you. Turns out there appears to be a 5 minute download timeout on the Update button from Home Assistant - Core. Over WiFi, I always hit that and have to plug into an Ethernet port.

hatchetaustralia, 2020-02-06:
Also same issue here

Proxmox VM, Supervisor 198, HA V 0.105.1

cheaptrick, 2020-02-06:
Having the same issue Oracle Virtualbox VM windows server 2012R2. I cannot install addons either getting the second error. 
 Supervisor version 198 and Host System 3.8 try to upgrade to 3.9

20-02-06 06:42:27 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32
20-02-06 06:47:59 INFO (SyncWorker_1) [hassio.docker.interface] Pull image hassioaddons/pi-hole-amd64 tag 4.0.0.
20-02-06 06:48:02 ERROR (SyncWorker_1) [hassio.docker.interface] Can't install hassioaddons/pi-hole-amd64:4.0.0 -> 500 Server Error: Internal Server Error ("readlink /mnt/data/docker/overlay2: invalid argument").

roybosch, 2020-02-06:
I'm running hassio too on an Intel NUC with Virtualbox. I'm also unable to update HassOS to version 3.9. The current version I'm running is 2.10.

I tried updating HassOS to 2.11 first, using the console. That gives the same errors:
`20-02-06 09:15:05 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.11/hassos_ova-2.11.raucb
20-02-06 09:15:14 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-2.11.raucb
20-02-06 09:15:14 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-02-06 09:15:14 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-02-06 09:15:15 INFO (MainThread) [hassio.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-02-06 09:15:15 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-02-06 09:15:15 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32`

pvizeli, 2020-02-06:
`readlink /mnt/data/docker/overlay2: invalid argument` is an docker issue with his overlay database. Run `hassio su repair` to rebuild the docker database/storage

roybosch, 2020-02-06:
I ran `hassio su repair`:

`20-02-06 10:33:17 INFO (MainThread) [hassio.core] Start repairing of Hass.io Environment
20-02-06 10:33:17 INFO (SyncWorker_0) [hassio.docker] Prune stale containers
20-02-06 10:33:17 INFO (SyncWorker_0) [hassio.docker] Prune stale images
20-02-06 10:33:24 INFO (SyncWorker_0) [hassio.docker] Prune stale builds
20-02-06 10:33:24 INFO (SyncWorker_0) [hassio.docker] Prune stale volumes
20-02-06 10:33:24 INFO (SyncWorker_0) [hassio.docker] Prune stale networks
20-02-06 10:33:24 INFO (MainThread) [hassio.addons] Found 3 add-ons to repair
20-02-06 10:33:24 INFO (MainThread) [hassio.addons] Start repair for add-on: core_letsencrypt
20-02-06 10:33:24 INFO (SyncWorker_1) [hassio.docker.interface] Pull image homeassistant/amd64-addon-letsencrypt tag 4.3.
20-02-06 10:33:40 INFO (MainThread) [hassio.addons] Start repair for add-on: core_check_config
20-02-06 10:33:40 INFO (SyncWorker_4) [hassio.docker.interface] Pull image homeassistant/amd64-addon-check_config tag 3.1.
20-02-06 10:34:39 INFO (MainThread) [hassio.addons] Start repair for add-on: a0d7b954_motioneye
20-02-06 10:34:39 INFO (SyncWorker_2) [hassio.docker.interface] Pull image hassioaddons/motioneye tag 0.6.0.
20-02-06 10:35:11 INFO (MainThread) [hassio.hassos] Repair HassOS CLI 15
20-02-06 10:35:11 INFO (SyncWorker_2) [hassio.docker.interface] Pull image homeassistant/amd64-hassio-cli tag 15.
20-02-06 10:35:12 INFO (SyncWorker_2) [hassio.docker.interface] Tag image homeassistant/amd64-hassio-cli with version 15 as latest
20-02-06 10:35:12 INFO (MainThread) [hassio.supervisor] Repair Supervisor 198
20-02-06 10:35:12 INFO (MainThread) [hassio.core] Finished repairing of Hass.io Environment`

Then I retried to update HassOS with `hassio os update`:

`20-02-06 10:36:24 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.9/hassos_ova-3.9.raucb
20-02-06 10:36:32 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-3.9.raucb
20-02-06 10:36:32 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-02-06 10:36:32 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-02-06 10:36:34 INFO (MainThread) [hassio.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-02-06 10:36:34 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-02-06 10:36:34 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32`

Same error :(

cheaptrick, 2020-02-06:
> `readlink /mnt/data/docker/overlay2: invalid argument` is an docker issue with his overlay database. Run `hassio su repair` to rebuild the docker database/storage

Thank you @pvizeli your solution worked for the addons but still not able to update operating system to 3.9. I appreciate that. Thank you

edwardscaleb9, 2020-02-07:
Same issue here.  Tried rebooting, updating from CLI, and hassio su repair.  No luck.  I'm running hassos 2.12 on ESXi.

andreskebe, 2020-02-07:
I was having the same issue, and did the same as @u17194907425 (removing adguard from my router and putting 8.8.8.8) and the update to the OS went through.

timdonovanuk, 2020-02-07:
Same error but with hassOS 3.9 on Proxmox:

```
20-02-07 10:17:43 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.9/hassos_ova-3.9.raucb
20-02-07 10:17:52 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-3.9.raucb
20-02-07 10:17:52 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-02-07 10:17:52 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-02-07 10:17:53 INFO (MainThread) [hassio.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-02-07 10:17:53 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-02-07 10:17:53 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32
```

timdonovanuk, 2020-02-07:
Update..I tried a second time and it worked. Bit uncomfortable with that, but ¯\_(ツ)_/¯

roybosch, 2020-02-07:
> Update..I tried a second time and it worked. Bit uncomfortable with that, but ¯\_(ツ)_/¯

Suddenly it worked here too. Everything's up to date. At first my front-end wouldn't load, but after rebuilding hassio, reloading the supervisor and rebooting the VM, everything works fine again.

jjmuriel, 2020-02-07:
Same, after a lot of reboots it is now working. Very weird

outrun0506, 2020-02-07:
Hmm, had the same issue. Seems like AdGuard Add-On cause this issue. After a disabled AdGuard, changed DNS-Server back to my router and rebooted HomeAssistant Host everything started working normal.

jjmuriel, 2020-02-07:
I don't have AdGuard Add-On and I had the same issue. I changed several times the DNS and nothing.  Only after several reboots I can updated HassOS.

edwardscaleb9, 2020-02-07:
Still no luck here.  I'm not using AdGuard, but I'm using PiHole.  I've checked query logs there and nothing is getting blocked.  I've rebooted several times and still can't update HassOS.

edwardscaleb9, 2020-02-07:
Weird... Rebooted one more time and I was able to upgrade.

monpelaud, 2020-02-08:
Hi,
Same issue for me too !
Can't upgrade hassos, I'm using homeassistant 0.104.3 and hassos: 3.7.
Is it possible to do an offline upgrade without install a new fresh hassio image?
Is there a tutorial for that?

Many thanks for your help.

Best regards

rjm12rjm12, 2020-02-08:
I also did some reboots and eventually the update went through.

Basanon, 2020-02-08:
"Operating System 3.9
You are currently running version 3.5"

Reboot did not help

```
WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.9/hassos_rpi4-3.9.raucb: Cannot connect to host github-production-release-asset-2e65be.s3.amazonaws.com:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')]
```

jensihnow, 2020-02-08:
This issue seems to be tricky, as I was able to update from 2.12 to 3.8 at some point. Interestingly, I'm currently running into the issue while updating to 3.9:

```
20-02-08 21:44:48 INFO (MainThread) [hassio.api.security] /supervisor/update access from a0d7b954_ssh
20-02-08 21:45:12 INFO (MainThread) [hassio.api.security] /hassos/info access from a0d7b954_ssh
20-02-08 21:45:17 INFO (MainThread) [hassio.api.security] /hassos/update access from a0d7b954_ssh
20-02-08 21:45:17 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.9/hassos_ova-3.9.raucb
20-02-08 21:45:33 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-3.9.raucb
20-02-08 21:45:33 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-02-08 21:45:33 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-02-08 21:45:34 INFO (MainThread) [hassio.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-02-08 21:45:34 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-02-08 21:45:34 ERROR (MainThread) [hassio.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32
```

Basanon, 2020-02-08:
Got same error message with _hassos update --log-level debug_
```
DEBU[0000] Adding homedir to searchpath                  homedir=/root
INFO[0000] No configfile found                          
DEBU[0000] Debug flags                                   apiToken=dc2074c88e18417bbc5b813c2c3979f1 cfgFile= endpoint=hassio logLevel=debug noProgress=false rawJSON=false
DEBU[0000] hassos update                                 args="[]"
DEBU[0000] [GenerateURI]                                 base=hassio command=update section=hassos
DEBU[0000] [GenerateURI] Result                          uri="http://hassio/hassos/update" url="http://hassio/hassos/update" url(string)="http://hassio/hassos/update"
⣽ Processing... DEBU[0001] Response                                      body="{\"result\": \"error\", \"message\": \"Unknown Error, see logs\"}" fields.time=1.206078735s headers="map[Content-Length:[57] Content-Type:[application/json; charset=utf-8] Date:[Sat, 08 Feb 2020 22:24:57 GMT] Server:[Python/3.7 aiohttp/3.6.1]]" recieved-at="2020-02-08 22:24:57.261137778 +0000 UTC m=+1.217687102" request="&{POST http://hassio/hassos/update HTTP/1.1 1 1 map[Accept:[application/json] Authorization:[Bearer dc2074c88e18417bbc5b813c2c3979f1] User-Agent:[go-resty/2.1.0 (https://github.com/go-resty/resty)] X-Hassio-Key:[dc2074c88e18417bbc5b813c2c3979f1]] <nil> 0x2d1c74 0 [] false hassio map[] map[] <nil> map[]   <nil> <nil> <nil> <nil>}" status="400 Bad Request" statuscode=400
Processing... Done.

Error: Unknown Error, see logs
```

Basanon, 2020-02-10:
Hassio, Pi4
Same with 3.5 -> 3.10 update
```
20-02-10 13:07:18 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/3.10/hassos_rpi4-3.10.raucb: Cannot connect to host github-production-release-asset-2e65be.s3.amazonaws.com:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')]
```

ddppddpp, 2020-02-10:
Trying to update hassos from 1.13 to 3.9 on a rpi3, both from Lovelace and from cli. Similar errors as above: Unknown Error, see logs.

ddppddpp, 2020-02-11:
Same story with 3.10. System log:
```
20-02-11 10:13:41 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/3.10/hassos_rpi3-3.10.raucb
20-02-11 10:18:41 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/3.10/hassos_rpi3-3.10.raucb: 
```
with no way to further trace the error :-(
Is there a possibility to do a 2-step update from 1.13?

fluxzero, 2020-02-12:
In the end, my issue seemed to be DNS related as well.  I was using the Dnsmasq add-on of HASS.IO as my primary DNS (to get my FQDN for Let's Encrypt working internally so I could use HTTPS on the mobile app properly).  Even though it was forwarding things to 8.8.8.8 appropriately, turning off Dnsmasq then trying to update the Hass OS again (thus presumably forcing the OS to failover to the secondary DNS) is what finally made it work.

ddppddpp, 2020-02-13:
I'm also running a split DNS setup, but stoping my local dns and defaulting to the ISP provided one didn't solve the update issue for me.

roybosch, 2020-02-13:
I didn't change anything to my DNS settings, and still the problem occured to me. After a day suddenly the error was gone. This seems to be totally random.

timdonovanuk, 2020-02-18:
I'm back! Just got the same error, uprading to the next version (3.10 - 3.11), where this is still a problem :)

`HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32`

TheWidowLicker, 2020-02-20:
Anyone working on this issue.
I changed my dns to 8.8.8.8, increased space on my vm, rebooted.
Still get same issue when I try to update from 3.8 to 3.9/3.10/3.11

Basanon, 2020-02-20:
Pi4
3.5->3.11 
```
20-02-20 18:52:18 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/3.11/hassos_rpi4-3.11.raucb: Cannot connect to host github-production-release-asset-2e65be.s3.amazonaws.com:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')]
```

monpelaud, 2020-02-22:
Hi,
After many attempts suddenly update 3.7 to 3.11 works ! Why ? I don't know.
Best regards

Basanon, 2020-02-27:
Updated from 3.5 to 3.11
Router blocked "amazonaws.com" domain for some reason.

toastyrhombus, 2020-03-06:
Been stuck for the latest few releases due to below error:
`HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32`
At the same time, I could not login to the Hassos shell (not the ssh addon shell) as upon logging in with root on the console to the ha cli and then running the 'login' command - the cli accepts the command, tells me cli.sh has exited with code 10 and then kicks me back out to the ha cli login prompt. Wasn't sure if related. Regardless, upgrading to 3.11 does not resolve this issue, separate issue probably.

Finally managed to update to 3.11 by downloading the raucb file and installing it with a USB stick as per  [https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md](url)

lassej01, 2020-03-07:
I have had this error twice for new releases. I'm running  on Windows10/Intel Nuc. Both times i have solved it by restarting the NUC.  (now on 3.12)

pvizeli, 2020-03-09:
Yeah, we use Rauc as update system. I hope the next release solve that.

obbardc, 2020-03-26:
relevant issue opened in Rauc: https://github.com/rauc/rauc/issues/562

closed 2020-05-08 by pvizeli

ddppddpp, 2020-05-25:
For whatever reason I keep getting error when trying to upgrade the os - both OTA and via USB import. Current setup - RPi3b, 32bit, os v. 1.13.
Any ideas how to complete an upgrade (besides the obvious clean isntall/restore)?

PR #659 (2020-05-08): Fix issue with loop devices
Fix: #544
Fix: #607