home-assistant/operating-system #545: Home Assistant not contactable after approx 24hrs using Hass.io on VirtualBox

stratplayer, 2020-01-15:
**Home Assistant release with the issue:**
v0.103.4 (and preceeding v0.103.x versions)

**Operating environment (HassOS/Generic):**
HassOS v3.7

**Supervisor logs:**
Not applicable here

**Description of problem:**
Recently there have been several of us who are experiencing an issue where Home Asisstant becomes unreachable from the UI using Hass.io running on Oracle VirtualBox v6.0.14r133895 (qt5.6.2) which is the latest installable version at the time of writing this.

After doing a "host reboot" to make the Home Assistant UI available, in less than about 24 hours the same problem occurs. This was repeated for over 2 weeks, I then had time to work with the experts on Discord to discuss this problem further.

When HA crashes my UI is not available and the none of the automations etc trigger, it is totally offline. The Host VM is ok and you can access the prompt to login as root etc. Performing a "ha restart" fixes nothing, only a "host reboot" makes the system work again (if only for about 24 hours).

Tinkerer mentioned he had seen other people mentioning the same problem and pointed me to this thread:
https://community.home-assistant.io/t/hassio-unstable-needs-to-be-sent-to-an-asylum/158247

Working with Tommii on the Discord he talked through some commands to check the IP config of the HassOS VM, I had been using reserved IP address via DHCP. This showed that no networking info existed when it was in a failed state, so HassOS had dropped off the network.

He then suggested to set the IP address manually in HassOS which I did using the following commands: 

> nmcli c mod $(nmcli -g uuid c) ipv4.method manual ipv4.addresses "192.168.0.189/24" ipv4.gateway "192.168.0.1" ipv4.dns "8.8.8.8,8.8.4.4"
> 
> nmcli c up $(nmcli -g uuid c)

This so far has been fully succesful, my Home Assistant instance has been available via the UI for over 72 hours now. Feedback from the originator of the above thread was also the same (his crashed every 24hrs too) but now is stable. Also today a newbie added to the thread saying that this looks to have cured his problem too.

Not sure what the root cause of the IP Address being dropped if via DHCP from router etc, but changing it to set it manually in HassOS seems to have cured the issue.

stratplayer, 2020-02-07:
Platform still functioning without issue. Applied all updates to HomeAssistant (currently 0.105.2 and still running without the same issue occuring)

Brianckramer, 2020-02-11:
I am having the same issue, HassOs running in Virtualbox on a Windows host.  The VM drops off the network every 24 hours.  Where do I run the terminal command?  It won't work on the HA-CLI, right?

stratplayer, 2020-02-11:
> I am having the same issue, HassOs running in Virtualbox on a Windows host. The VM drops off the network every 24 hours. Where do I run the terminal command? It won't work on the HA-CLI, right?

Yes login to host as "root" to access the system. Then type "login", this will take you to the environment to apply to commands above. 

It has worked perfectly for the group of us who followed the steps

Brianckramer, 2020-02-12:
Okay thanks, I will report back.

pvizeli, 2020-02-13:
Is a bug they affect on VirtualBox, other devices/vm they I run work fine

Brianckramer, 2020-02-14:
Confirm the fix does work, up for 1.6 days now.  I wouldn't be surprised if it VirtualBoxe's fault, just glad I found something that works.  HyperV doesn't allow USB passthrough for my Zwave stick, and I would love to ditch windows but I need it for Blue Iris.  Thanks @stratplayer .

stratplayer, 2020-02-14:
Great news. Didn't matter how we achieved it, but we have stability

nhorvath, 2020-02-24:
I had this issue last time I upgraded hass os and now I upgraded again and I saw this similar issue so I want to add info here. After upgrading it seems the system gets a dhcp lease before the clock is properly set and you wind up with a lease that expires many months from now. So it never renews the lease and drops off the network.

You can see below that I have lease files from this coming november. Last time this happened, removing the bad leases and restarting fixed the issue. Hoping that works again.

[image]

nhorvath, 2020-02-24:
Wanted to clarify my last comment in case it helps someone else. I mean restarting network manager after removing the bad lease file, not the vm. 
```
systemctl restart NetworkManager
```

stratplayer, 2020-02-24:
That’s great info. Hopefully it can be used to eradicate the issue long term.

bwthor, 2020-03-15:
@stratplayer and @nhorvath.  Many thanks for finding these issues.  I have been having issues for way too long with my network dropping unexpectedly.  I resorted to an automation to reboot the host every 24 hours, which was a successful patch to my issue.  I set the IP statically using the info above, and have been running for a week now with no issues!

stratplayer, 2020-03-15:
Fantastic news, it was driving me mad when it kept happening.

pvizeli, 2020-05-08:
Fix with 4.6

closed 2020-05-08 by pvizeli

nschwermann, 2021-06-29:
Having this issue running virtual box on OSX. I lose internet connection every hour or so, only way to get it back is put in the host reboot command.

stratplayer, 2021-06-29:
Did you perform the steps provided that fixes it? Set it to a reserved IP address in DHCP first then do steps below. Never had issue since.

nmcli c mod $(nmcli -g uuid c) ipv4.method manual ipv4.addresses "192.168.20.170/24" ipv4.gateway "192.168.20.1" ipv4.dns "8.8.8.8,8.8.4.4"

edit IP addresses for your situation

nmcli c up $(nmcli -g uuid c)

nschwermann, 2021-06-29:
When I tried I got an error that said "unknown command "nmcli" for "ha"

stratplayer, 2021-06-29:
You need to go a level deeper.

After you login with “root”

Type “login”

That should take you to the system prompt for those commands.

nschwermann, 2021-06-29:
Thanks, that seemed to halfway fix it. Now I can reach from local host address on my mac but still net info shows no ip and no internet address and I cant connect from my phone.

nhorvath, 2021-06-29:
Delete your dhcp lease files in ha cli and reboot the vm.

nschwermann, 2021-06-30:
Thanks everything is working now!

nschwermann, 2021-06-30:
@nhorvath one final issue here my supervisor still does not have internet.

nhorvath, 2021-06-30:
Sorry, I don't know why that would be. The main issue happens because after
running for a long time the click drifts in the VM and if there is a
restart and it gets a new lease the expiration date gets to be in the
future and the lease doesn't renew so it falls off the network. ntp doesn't
update the clock until after the network is up, for obvious reasons.

Have you tried restarting the supervisor since fixing the main lease? I'm
not sure if the supervisor has its own file that you can clear.

nschwermann, 2021-06-30:
Yep I tried to restart the supervisor, could the issue be I put the wrong gateway?  In my router settings I seem to have two gateways, IPv4 default gateway and DHCP Server default gateway, they are different addresses.

nhorvath, 2021-06-30:
The default gateway should be the lan address of your router. Your DHCP
server shouldn't be pushing itself as default gw unless it has a wan
interface on the internet.

nschwermann, 2021-06-30:
Hmm ok based on logs it can't reach some IP address under the docker section in `net info` I tried trace routing them from my mac and it can't seem to reach the endpoint either 172.30.32.2 so seems to be unrelated to the settings you gave to fix my local connection.

bigshow104, 2022-04-23:
> You need to go a level deeper. After you login with “root” Type “login” That should take you to the system prompt for those commands.
> […](#)

please can you tell me how i log in as root,

kaspars1985, 2023-05-20:
> Hi I am trying to solve this problem but having issues (feel free to treat me as the idiot I am!)
> At the command, I am typing root, and then my root password, then I am typing the command above, but getting an error:
> I can't work out what I am doing wrong. I tried typing "login" after I had logged in as root - but then it gives me 60 seconds and times me out from typing the command.
> I am using a Mac with Debian running on UTM.
> Any help would be amazing because I am pulling my hair out!
> Matt

Hi!

I am honestly noob myself :))
I remember I got it working ans since there hasnt been any more shutdowns. I remember i tried search on google as well

Ar cieņu,

Kaspars Ciematnieks

srescio, 2024-02-01:
same issue 🙃