home-assistant/operating-system #546: CC2531 working on rPi but not on OVA image on proxmox

Iam-lcl, 2020-01-19:
**HassOS release with the issue:**

3.8 and dev 4.0

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

```
[    0.569713] ACPI: bus type USB registered
[    0.569713] usbcore: registered new interface driver usbfs
[    0.569713] usbcore: registered new interface driver hub
[    0.569713] usbcore: registered new device driver usb
[    0.846045] pci 0000:00:01.2: quirk_usb_early_handoff+0x0/0x660 took 63830 usecs
[    0.910851] pci 0000:00:1d.0: quirk_usb_early_handoff+0x0/0x660 took 63121 usecs
[    0.975840] pci 0000:00:1d.1: quirk_usb_early_handoff+0x0/0x660 took 63429 usecs
[    1.040575] pci 0000:00:1d.2: quirk_usb_early_handoff+0x0/0x660 took 63175 usecs
[    1.105311] pci 0000:00:1d.7: quirk_usb_early_handoff+0x0/0x660 took 63180 usecs
[    1.202554] pci 0000:01:1b.0: quirk_usb_early_handoff+0x0/0x660 took 94894 usecs
[    1.394981] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.429782] ehci-pci 0000:00:1d.7: new USB bus registered, assigned bus number 1
[    1.438357] ehci-pci 0000:00:1d.7: USB 2.0 started, EHCI 1.00
[    1.439087] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[    1.439765] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.440439] usb usb1: Product: EHCI Host Controller
[    1.441070] usb usb1: Manufacturer: Linux 5.4.11 ehci_hcd
[    1.441717] usb usb1: SerialNumber: 0000:00:1d.7
[    1.442519] hub 1-0:1.0: USB hub found
[    1.444322] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.445728] uhci_hcd: USB Universal Host Controller Interface driver
[    1.480615] uhci_hcd 0000:00:01.2: new USB bus registered, assigned bus number 2
[    1.482548] usb usb2: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.04
[    1.483351] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.484132] usb usb2: Product: UHCI Host Controller
[    1.484949] usb usb2: Manufacturer: Linux 5.4.11 uhci_hcd
[    1.485842] usb usb2: SerialNumber: 0000:00:01.2
[    1.486822] hub 2-0:1.0: USB hub found
[    1.522923] uhci_hcd 0000:00:1d.0: new USB bus registered, assigned bus number 3
[    1.525078] usb usb3: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.04
[    1.525904] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.526735] usb usb3: Product: UHCI Host Controller
[    1.527543] usb usb3: Manufacturer: Linux 5.4.11 uhci_hcd
[    1.528349] usb usb3: SerialNumber: 0000:00:1d.0
[    1.529261] hub 3-0:1.0: USB hub found
[    1.565133] uhci_hcd 0000:00:1d.1: new USB bus registered, assigned bus number 4
[    1.566985] usb usb4: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.04
[    1.567787] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.568581] usb usb4: Product: UHCI Host Controller
[    1.569378] usb usb4: Manufacturer: Linux 5.4.11 uhci_hcd
[    1.570135] usb usb4: SerialNumber: 0000:00:1d.1
[    1.571068] hub 4-0:1.0: USB hub found
[    1.607416] uhci_hcd 0000:00:1d.2: new USB bus registered, assigned bus number 5
[    1.609230] usb usb5: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.04
[    1.610036] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.610861] usb usb5: Product: UHCI Host Controller
[    1.611785] usb usb5: Manufacturer: Linux 5.4.11 uhci_hcd
[    1.612571] usb usb5: SerialNumber: 0000:00:1d.2
[    1.613483] hub 5-0:1.0: USB hub found
[    1.650354] xhci_hcd 0000:01:1b.0: new USB bus registered, assigned bus number 6
[    1.656731] usb usb6: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[    1.657553] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.658376] usb usb6: Product: xHCI Host Controller
[    1.659336] usb usb6: Manufacturer: Linux 5.4.11 xhci-hcd
[    1.660128] usb usb6: SerialNumber: 0000:01:1b.0
[    1.661114] hub 6-0:1.0: USB hub found
[    1.664400] xhci_hcd 0000:01:1b.0: new USB bus registered, assigned bus number 7
[    1.665171] xhci_hcd 0000:01:1b.0: Host supports USB 3.0 SuperSpeed
[    1.666116] usb usb7: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.666989] usb usb7: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.04
[    1.667823] usb usb7: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.668652] usb usb7: Product: xHCI Host Controller
[    1.669603] usb usb7: Manufacturer: Linux 5.4.11 xhci-hcd
[    1.670427] usb usb7: SerialNumber: 0000:01:1b.0
[    1.671408] hub 7-0:1.0: USB hub found
[    1.673723] usbcore: registered new interface driver usblp
[    1.674753] usbcore: registered new interface driver usb-storage
[    1.675588] usbcore: registered new interface driver usbserial_generic
[    1.676372] usbserial: USB Serial support registered for generic
[    1.696095] usbcore: registered new interface driver usbhid
[    1.698317] usbhid: USB HID core driver
[    1.823470] usb 2-1: new full-speed USB device number 2 using uhci_hcd
[    2.115482] usb 3-1: new full-speed USB device number 2 using uhci_hcd
[    2.139682] usb 2-1: not running at top speed; connect to a high speed hub
[    2.173250] usb 2-1: New USB device found, idVendor=0627, idProduct=0001, bcdDevice= 0.00
[    2.174395] usb 2-1: New USB device strings: Mfr=1, Product=3, SerialNumber=10
[    2.175294] usb 2-1: Product: QEMU USB Tablet
[    2.176131] usb 2-1: Manufacturer: QEMU
[    2.176988] usb 2-1: SerialNumber: 28754-0000:00:01.2-1
[    2.178334] usb 6-1: new high-speed USB device number 2 using xhci_hcd
[    2.186672] input: QEMU QEMU USB Tablet as /devices/pci0000:00/0000:00:01.2/usb2/2-1/2-1:1.0/0003:0627:0001.0001/input/input4
[    2.189198] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:01.2-1/input0
[    2.202342] usb 6-1: New USB device found, idVendor=090c, idProduct=1000, bcdDevice=11.00
[    2.203422] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.204425] usb 6-1: Product: Flash Drive
[    2.205355] usb 6-1: Manufacturer: Samsung
[    2.206228] usb 6-1: SerialNumber: 0363015100009690
[    2.208638] usb-storage 6-1:1.0: USB Mass Storage device detected
[    2.210874] scsi host8: usb-storage 6-1:1.0
[    2.814965] usb 3-1: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=39.44
[    2.816188] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.817125] usb 3-1: Product: CC2531 USB Dongle
[    2.818036] usb 3-1: Manufacturer: Texas Instruments
[    2.894480] usb 6-2: new low-speed USB device number 3 using xhci_hcd
[    3.032681] usb 6-2: New USB device found, idVendor=413c, idProduct=2107, bcdDevice= 1.78
[    3.033688] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.034710] usb 6-2: Product: Dell USB Entry Keyboard
[    3.035676] usb 6-2: Manufacturer: Dell
[    3.042669] input: Dell Dell USB Entry Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:01:1b.0/usb6/6-2/6-2:1.0/0003:413C:2107.0002/input/input5
[    3.097353] hid-generic 0003:413C:2107.0002: input,hidraw1: USB HID v1.10 Keyboard [Dell Dell USB Entry Keyboard] on usb-0000:01:1b.0-2/input0
```
```
[    2.814965] usb 3-1: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=39.44
[    2.816188] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.817125] usb 3-1: Product: CC2531 USB Dongle
[    2.818036] usb 3-1: Manufacturer: Texas Instruments
[    2.894480] usb 6-2: new low-speed USB device number 3 using xhci_hcd
[    3.032681] usb 6-2: New USB device found, idVendor=413c, idProduct=2107, bcdDevice= 1.78
[    3.033688] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.034710] usb 6-2: Product: Dell USB Entry Keyboard
[    3.035676] usb 6-2: Manufacturer: Dell
[    3.042669] input: Dell Dell USB Entry Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:01:1b.0/usb6/6-2/6-2:1.0/0003:413C:2107.0002/input/input5
[    3.097353] hid-generic 0003:413C:2107.0002: input,hidraw1: USB HID v1.10 Keyboard [Dell Dell USB Entry Keyboard] on usb-0000:01:1b.0-2/input0
```
**Description of problem:**
I have e CC2531 zigbee usb dongle, that if connected into the rPi installation running HassOS 3.8 it shows up under hardware:
[image]

Connecting this same usb into my NUC running proxmox with hass.io (ova image) installed with https://github.com/whiskerz007/proxmox_hassos_install
And usb hardware added to the VM, the device is not shown under hardware at all.
SSH and running lsusb show the device there, also the dmesg shows the usb, but does not look like it is installed?
[image]

I think there is some missing drivers for CC2531 that exists in rPi image but not in ova image.

I have mapped the dell keyboard to make sure it is recognized by the OS and that does, so I think the proxmox usb hardware mapping is working correctly to hassOS just that some drivers are missing on HassOS.

pvizeli, 2020-01-20:
try:
`hassio host update --version 4.0`

The driver should be backported to 3.9 on  next release.

Iam-lcl, 2020-01-20:
did you mean `hassio os update --version 4.0` ?
`hassio host update --version 4.0`

Did not work, but anyway os update did't do any difference:
```
[    2.918945] usb 3-1: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=39.44
[    2.920146] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.921245] usb 3-1: Product: CC2531 USB Dongle
[    2.922167] usb 3-1: Manufacturer: Texas Instruments
[    2.996494] usb 6-2: new low-speed USB device number 3 using xhci_hcd
[    3.133970] usb 6-2: New USB device found, idVendor=413c, idProduct=2107, bcdDevice= 1.78
[    3.134960] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.135897] usb 6-2: Product: Dell USB Entry Keyboard
[    3.136865] usb 6-2: Manufacturer: Dell
[    3.143311] input: Dell Dell USB Entry Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:01:1b.0/usb6/6-2/6-2:1.0/0003:413C:2107.0002/input/input5
[    3.197598] hid-generic 0003:413C:2107.0002: input,hidraw1: USB HID v1.10 Keyboard [Dell Dell USB Entry Keyboard] on usb-0000:01:1b.0-2/input0

```

`hassio hardware info`:
```
audio: {}
disk: []
gpio: []
input:
- Dell Dell USB Entry Keyboard
- ImExPS/2 Generic Explorer Mouse
- Power Button
- QEMU QEMU USB Tablet
- AT Translated Set 2 keyboard
serial:
- /dev/ttyS0
- /dev/ttyS2
- /dev/ttyS3
- /dev/ttyS1

```

hassio host info:
```
chassis: vm
cpe: cpe:2.3:o:home_assistant:hassos:4.0:*:development:*:*:*:ova:*
deployment: development
features:
- reboot
- shutdown
- services
- hostname
- hassos
hostname: hassio
kernel: 5.4.11
operating_system: HassOS 4.0
```

pvizeli, 2020-01-20:
Which kernel modules are loaded?

Iam-lcl, 2020-01-20:
How do I see that?

SalkinDK, 2020-01-21:
@Iam-lcl I have my CC2531 working on a proxmox hassio install using the same guide.
How did you add the usb to the VM?

I did it like this, maybe that helps you?
[image]

[image]

Iam-lcl, 2020-01-21:
Yes, that is how I have it added
[image]
Mine does not say TI nor CDC... and only differ on the last ID 16ae on mine vs 16a8 on yours.

What HassOS version are you running?

SalkinDK, 2020-01-21:
Latest stable:

System | HassOS 3.8
Deployment | production

But it was working before i updated as well

Iam-lcl, 2020-01-21:
Yes, have added it when off.

One question, as I have seen people flashing the dongle, is that needed?
Have you flashed the cc2531 or is it running native firmware?

SalkinDK, 2020-01-21:
Yes i have flashed the zigbee2mqtt firmware onto it, otherwise it doesnt work at all?!

So you didnt had a working setup on your pi?

Iam-lcl, 2020-01-21:
Ah, I see, then this is probably it then?
does't explain why it's shown on the pi but not on the proxmox nuc. it's shown even in windows (but missing drivers)
I probably need to buy me a conbee II as I dont use mqtt.

pvizeli, 2020-01-22:
The serial driver is available. Maybe he needs a different Linux driver with the modified firmware. However, I can only fix that if someone can exactly saying what is need on the Linux kernel to support it.

SalkinDK, 2020-01-23:
as i wrote mine works fine when flashed with the zigbee2mqtt firmware.
So i think the driver is just fine.
I can't see no reason to plug the cc2531 usb stick without flashing firmware on it first.

closed 2020-02-03 by pvizeli