home-assistant/operating-system #548: Intel AC-7265 wireless/bluetooth problem

alesoft73, 2020-01-20:
I have Installing Hass.io with intel-nuc image 3.8
The intel AC-7265 wireless/bluetooth work a few minuts. After don't work. (bluetooth)
Is possible to update the driver ?
Many thanks!

pvizeli, 2020-01-23:
This it work better with 4.0? `hassio os update --version 4.0`

alesoft73, 2020-01-23:
> 
> 
> This it work better with 4.0? `hassio os update --version 4.0`

I test, many thanks!

alesoft73, 2020-01-23:
Work a few minuts... After down
shiw image
[image]

alesoft73, 2020-01-23:
[image]

alesoft73, 2020-02-10:
Many thanks, in version 3.10 bluetooth work perfect.
I love you :-)

alesoft73, 2020-02-10:
> 
> 
> This it work better with 4.0? `hassio os update --version 4.0`

Many thanks, in version 3.10 bluetooth work perfect.
And  have test with bluetootrh dongle with broadcom! perfect too!
I love you :-)

closed 2020-02-10 by pvizeli