home-assistant/operating-system #549: ODROID N2 no boot on Hassos 4.0

pepeEL, 2020-01-21:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Hi
Today i buy odroid N2 and also flash SPI Boot Manager name Petiboot. Then i flash on USB SSD and also on SDCard  image by BalenaEtcher Hassos 4.0. WHen insert it to Odroid N2 and try boot i can not. In boot manager i try select disk with Hassos but on USB i can not found , but on SDcard i can select but can not save seetings in Petiboot. For my opinion HASSOS 4.0 not working and now not support Odroid N2.
Please any help me....

pvizeli, 2020-01-22:
https://forum.odroid.com/viewtopic.php?f=54&t=37392&p=278586#p278586

pepeEL, 2020-01-23:
INFO. Only no boot from USB. 
From eMMC i no test, but from SDcard works ok but must select switch to MMC(not SPI),

Multipage, 2020-02-28:
I'm very interested in this too.  I have a 500GB SSD and Odroid N2 laying around that are useless.
If anyone has some tips on how to create a bootable USB SSD with hassio/homeassistant i'm all ears!
Don't want an ubuntu or armbian installation.

pepeEL, 2020-03-08:
Hi. Any news about implement it ?

closed 2020-05-08 by pvizeli