home-assistant/operating-system #551: Random MAC address at reboot on Odroid N2

Twanislas, 2020-01-23:
Hi !

Adding @pvizeli for viz, it looks you're working on the Odriod N2 support...

**HassOS release with the issue:**
```
$ hassio hassos info
hassio hassos info
board: odroid-n2
boot: A
version: "4.0"
version_cli: "15"
version_cli_latest: "15"
version_latest: "4.0"
```

**Description of problem:**
At every reboot, the Odroid box is getting a new (random ?) mac address.
Some digging yields that there may be a parameter to set in some boot config files : https://forum.libreelec.tv/thread/2308-odroid-c2-7-90-008-mac-address-changes-randomly-after-reboot/?postID=18264#post18264

Let me know if I can be of any help :)

Thanks !

pepeEL, 2020-01-23:
Confirm it. Also i have the same problem. Every reboot Hassos generate other mac and router give other IP. This mac is other than on label odroid n2.
Please resolved this problem... This is big bug.

pepeEL, 2020-01-27:
Any news when it was resolved ?

pepeEL, 2020-01-28:
Or any temporary solution ?

pvizeli, 2020-01-28:
Add a fixed IP over NetworkManager like in the docs

pepeEL, 2020-01-28:
In NetworkManager ? Can you tell me how or give me link to manual ?
Please

pepeEL, 2020-01-28:
And is any chance to add in image 4.1 for odroid N2 add option to boot from USB ?

pvizeli, 2020-01-28:
@pepeEL stop to make cross-postings or flood issues. That makes it very hard to follow productively with people they work on a solution. I updated the post above with the link...

Twanislas, 2020-01-28:
> I try to fix that with the next 4.1update in maybe 2-3 weeks.
Thanks a bunch Pascal ! :)

pvizeli, 2020-01-31:
okay, the fix work. We can read the origin mac address and set that once. I need check C2/XU and we can merge the fix.

closed 2020-02-01 by pvizeli

Twanislas, 2020-02-04:
Thanks Pascal ! :)

pepeEL, 2020-02-04:
Also very thanks

Twanislas, 2020-02-12:
Hi !

It looks like it did not fix the issue for me as I still face the problem...
Does it need to be a full reinstall and would not work for an update from 4.0 to 4.2 ?

How can I check the logs of the oneshot script ?

Thanks !!

pvizeli, 2020-02-13:
You need reflash your device or wait until 4.3 is out with a fix on OTA bootloader update

Twanislas, 2020-02-13:
Oh I see. It can wait until 4.3 is out, I've got a short lease time on my DHCP server and I don't reboot that often :D
Thanks for the precision ;)

Twanislas, 2020-03-02:
FWIW, this works allright now :)
Thanks !

PR #556 (2020-02-01): Fix odroid issues
Fix: #551