home-assistant/operating-system #554: Add option Time Zone

pepeEL, 2020-01-28:
Hi
I use HASSOS and i have a question... Is any chance to add option setup Time Zone ? In logs is a problem that show other time that i have in my time zone. Option setup time zone will be a good option...

pvizeli, 2020-01-28:
Like in all previous requests, the OS runs like a supervisor on UTC. You can set up in Home Assistant a timezone and they are replication to all containers they run on top of the system.

Feel free to add the systemd time service dbus support to the supervisor that the supervisor can change the OS timezone if you like. That will be the only allowed way to support this request. Please close this issue if you don't want to look into and try to provide PR.

pepeEL, 2020-01-28:
On please explaine me or give me link how i can change it... I am beginner user sorry.

pepeEL, 2020-01-28:
You think about it ?
homeassistant:
  time_zone: America/Los_Angeles

pepeEL, 2020-01-28:
I have setup time zone in Home Assistant as Europ/Warswa but still in hass.io>System>System Log   i have 1 hour diffrent... Should show me log with time 17:xx:xx but show me 16:xx:xx

Ysbrand, 2020-02-19:
Well, I guess it makes sense to run the OS in UTC, but somehow timestamps in the logs visible in the GUI will need an option to be translated into whatever local time.

pvizeli, 2020-02-19:
> Feel free to add the systemd time service dbus support to the supervisor, so that the supervisor can change the OS timezone if you like.

closed 2020-04-26 by stale[bot]