home-assistant/operating-system #555: Attaching external usb drive fails

bitboxer, 2020-01-29:
**HassOS release with the issue:**

0.104.3

**Description of problem:**

When I attach an external disk to the system it fails to mount the disk with this output:

```
[16745144.094377] usb 1-1.3.1: USB disconnect, device number 8
[16745144.104699] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[16745144.108628] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[16745146.751714] usb 1-1.3.1: new high-speed USB device number 9 using dwc_otg
[16745146.883976] usb 1-1.3.1: New USB device found, idVendor=0480, idProduct=0900
[16745146.887028] usb 1-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16745146.892886] usb 1-1.3.1: Product: External USB 3.0
[16745146.895784] usb 1-1.3.1: Manufacturer: TOSHIBA
[16745146.898569] usb 1-1.3.1: SerialNumber: 20191120007562F
[16745146.904454] usb-storage 1-1.3.1:1.0: USB Mass Storage device detected
[16745146.907834] scsi host0: usb-storage 1-1.3.1:1.0
[16745149.718103] scsi 0:0:0:0: Direct-Access     TOSHIBA  External USB 3.0 5438 PQ: 0 ANSI: 6
[16745149.726717] sd 0:0:0:0: [sda] 1953525164 512-byte logical blocks: (1.00 TB/932 GiB)
[16745149.734136] sd 0:0:0:0: [sda] Write Protect is off
[16745149.737136] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
[16745149.738549] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[16745149.755670]  sda: sda1
[16745149.760986] sd 0:0:0:0: [sda] Attached SCSI disk
[16745149.764379] udevd[5483]: inotify_add_watch(6, /dev/sda, 10) failed: No such file or directory
[16745149.770866] udevd[5483]: inotify_add_watch(6, /dev/sda1, 10) failed: No such file or directory
```

pvizeli, 2020-01-29:
I can't see an issue. The `inotify_add_watch` is a known udev thing because of raid kernel module and can be ignore. You can't also mount an USB drive. What do you except?

bitboxer, 2020-01-30:
If it's okay that I cannot mount an usb drive, I will close this and will have to switch to another linux distro and install home assistant core to that one.

closed 2020-01-30 by bitboxer