home-assistant/operating-system #558: hasos on odroid xu4

MattL0, 2020-01-31:
is is working on emmc? Nothing is booting.

MattL0, 2020-01-31:
I tried different version on emmc nothing is booting

pvizeli, 2020-01-31:
XU4 is very complex for booting with eMMC unlike the C2/N2 images they work on SD like on eMMC. See at: https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md

Joentje, 2020-02-01:
I'm having the same issues. Flashed HassOS 3.8 and 4.0. Same result. After flashing 4.0 it was complaining the checksum was not matching in Etcher.

I followed all the steps from: https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md

When I flashed HassOS on a SD card and the emmc, and booted from SD. The `/dev/mmcblk0` did not exists. So the I used the `dd if=/dev/mmcblk1 of=/dev/mmcblk0boot0 bs=512 skip=1 seek=0 count=16381` command. That didn't work either.

After flashing Ubuntu, it is just working fine.

So no I'm a bit lost :)

collse, 2020-02-11:
@Joentje you need to use the following:
1) echo 0 > /sys/block/mmcblk0boot0/force_ro
3) dd if=/dev/mmcblk0 of=/dev/mmcblk0boot0 bs=512 skip=63 seek=62 count=1440
3) echo 1 > /sys/block/mmcblk0boot0/force_ro

closed 2020-04-18 by stale[bot]