home-assistant/operating-system #559: Z-Wave Raspberry board (RaZberry 2) and HassOS

artomka, 2020-01-31:
**HassOS release with the issue:**
HassOS 3.8

**Supervisor logs:**
```
20-01-31 16:21:15 INFO (MainThread) [__main__] Initialize Hass.io setup
20-01-31 16:21:16 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to Super                                                                 visor homeassistant/aarch64-hassio-supervisor with version 195
20-01-31 16:21:16 INFO (MainThread) [__main__] Setup HassIO
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.fr                                                                 eedesktop.systemd1 - /org/freedesktop/systemd1
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.fr                                                                 eedesktop.hostname1 - /org/freedesktop/hostname1
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: de.pen                                                                 gutronix.rauc - /
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.fr                                                                 eedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager
20-01-31 16:21:16 INFO (MainThread) [hassio.host.info] Update local host informa                                                                 tion
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DB                                                                 us.Properties.GetAll on /org/freedesktop/hostname1
20-01-31 16:21:16 INFO (MainThread) [hassio.host.services] Update service inform                                                                 ation
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.sy                                                                 stemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-01-31 16:21:16 INFO (MainThread) [hassio.host.network] Update local network D                                                                 NS information
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DB                                                                 us.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-01-31 16:21:16 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profile                                                                 s: {'hassio-supervisor'}
20-01-31 16:21:16 INFO (MainThread) [hassio.host.services] Reload local service                                                                  hassos-apparmor.service
20-01-31 16:21:16 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.sy                                                                 stemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
20-01-31 16:21:16 INFO (SyncWorker_2) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-hassio-dns with version 1
20-01-31 16:21:16 INFO (MainThread) [hassio.dns] Start CoreDNS plugin
20-01-31 16:21:16 INFO (MainThread) [hassio.misc.forwarder] Start DNS port forwa                                                                 rding to 172.30.32.3
20-01-31 16:21:16 INFO (SyncWorker_4) [hassio.docker.interface] Clean hassio_dns                                                                  application
20-01-31 16:21:17 INFO (SyncWorker_4) [hassio.docker.dns] Start DNS homeassistan                                                                 t/aarch64-hassio-dns with version 1 - 172.30.32.3
20-01-31 16:21:17 INFO (SyncWorker_1) [hassio.docker.interface] Attach to homeas                                                                 sistant/raspberrypi4-64-homeassistant with version 0.104.3
20-01-31 16:21:18 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DB                                                                 us.Properties.GetAll on /
20-01-31 16:21:18 INFO (MainThread) [hassio.hassos] Detect HassOS 3.8 / BootSlot                                                                  A
20-01-31 16:21:18 INFO (SyncWorker_0) [hassio.docker.hassos_cli] Found HassOS CL                                                                 I homeassistant/aarch64-hassio-cli with version 15
20-01-31 16:21:18 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/                                                                 core repository
20-01-31 16:21:18 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/                                                                 git/a0d7b954 repository
20-01-31 16:21:18 INFO (MainThread) [hassio.store] Load add-ons from store: 63 a                                                                 ll - 63 new - 0 remove
20-01-31 16:21:18 INFO (MainThread) [hassio.addons] Found 11 installed add-ons
20-01-31 16:21:19 INFO (SyncWorker_3) [hassio.docker.interface] Attach to hassio                                                                 addons/motioneye with version 0.6.0
20-01-31 16:21:19 INFO (SyncWorker_0) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-nginx_proxy with version 2.6
20-01-31 16:21:19 INFO (SyncWorker_2) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-configurator with version 4.2
20-01-31 16:21:19 INFO (SyncWorker_5) [hassio.docker.interface] Attach to hassio                                                                 addons/ide-aarch64 with version 2.2.0
20-01-31 16:21:19 INFO (SyncWorker_8) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-duckdns with version 1.10
20-01-31 16:21:19 INFO (SyncWorker_9) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-mosquitto with version 5.1
20-01-31 16:21:19 INFO (SyncWorker_3) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-ssh with version 8.1
20-01-31 16:21:19 INFO (SyncWorker_10) [hassio.docker.interface] Attach to homea                                                                 ssistant/aarch64-addon-check_config with version 3.1
20-01-31 16:21:19 INFO (SyncWorker_2) [hassio.docker.interface] Attach to homeas                                                                 sistant/aarch64-addon-samba with version 9.0
20-01-31 16:21:19 INFO (SyncWorker_7) [hassio.docker.interface] Attach to hassio                                                                 addons/sonweb-aarch64 with version 0.9.0
20-01-31 16:21:19 INFO (SyncWorker_12) [hassio.docker.interface] Attach to hassi                                                                 oaddons/ssh-aarch64 with version 7.0.1
20-01-31 16:21:19 INFO (MainThread) [hassio.updater] Fetch update data from http                                                                 s://version.home-assistant.io/stable.json
20-01-31 16:21:19 INFO (MainThread) [hassio.snapshots] Found 2 snapshot files
20-01-31 16:21:19 INFO (MainThread) [hassio.discovery] Load 1 messages
20-01-31 16:21:19 INFO (MainThread) [hassio.ingress] Load 0 ingress session
20-01-31 16:21:19 INFO (MainThread) [hassio.secrets] Load Home Assistant secrets                                                                 : 27
20-01-31 16:21:19 INFO (MainThread) [__main__] Run Hass.io
20-01-31 16:21:19 INFO (MainThread) [hassio.api] Start API on 172.30.32.2
20-01-31 16:21:19 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rau                                                                 c.Installer.Mark on /
20-01-31 16:21:19 INFO (MainThread) [hassio.hassos] Rauc: A - marked slot kernel                                                                 .0 as good
20-01-31 16:21:19 INFO (MainThread) [hassio.addons] Phase 'initialize' start 0 a                                                                 dd-ons
20-01-31 16:21:19 INFO (MainThread) [hassio.addons] Phase 'system' start 2 add-o                                                                 ns
20-01-31 16:21:19 INFO (SyncWorker_0) [hassio.docker.interface] Clean addon_a0d7                                                                 b954_sonweb application
[... cut ...]
2020-01-31 17:26:15.878 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2020-01-31 17:26:15.878 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2020-01-31 17:26:15.878 Detail, 
2020-01-31 17:26:15.878 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-01-31 17:26:15.879 Detail, contrlr, Notification: DriverFailed
```

pvizeli, 2020-02-02:
Sounds like `RaZberry 2` use GPIO or so. You can disable bluetooth with `config.txt` and `dtoverlay=pi3-disable-bt`. I used that before I switch to USB hardware from GPIO and it work.

artomka, 2020-02-03:
@pvizeli Hi, thanks for the advice. However, it still does not work. I have changed `dtoverlay=pi3-miniuart-bt` to `dtoverlay=pi3-disable-bt` in the config file, reboot and in the HA configuration file I have: 
```
zwave:
  usb_path: /dev/ttyAMA0
```
Unfortunately, it still does not work :( 
```
2020-02-03 09:37:48.038 Always, OpenZwave Version 1.4.3452 Starting Up
2020-02-03 09:38:03.235 Info, Setting Up Provided Network Key for Secure Communications
2020-02-03 09:38:03.235 Warning, Failed - Network Key Not Set
2020-02-03 09:38:03.235 Info, mgr, Added driver for controller /dev/ttyAMA0
2020-02-03 09:38:03.235 Info,   Opening controller /dev/ttyAMA0
2020-02-03 09:38:03.235 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)
2020-02-03 09:38:03.235 Info, Serial port /dev/ttyAMA0 opened (attempt 1)
2020-02-03 09:38:03.236 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-02-03 09:38:03.236 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2020-02-03 09:38:03.236 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2020-02-03 09:38:03.236 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2020-02-03 09:38:03.236 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2020-02-03 09:38:03.237 Detail, 
2020-02-03 09:38:03.237 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-02-03 09:38:03.239 Detail, contrlr, Notification: DriverFailed
```

artomka, 2020-02-04:
So, finally, I sold the board. It was quite uncomfortable to use it without the ability to use Bluetooth module, which is really useful. Even so, thank you for your time.

closed 2020-04-11 by stale[bot]

carryonrewardless, 2020-04-24:
Can this be re-opened. 
My Pi4 using a Razberry on the GPIO needed this mod to config.txt when setting up a new Home Assistant installation. 
Using 32 bit HassIO 0.108.8 on kernel 4.19.114-v7l.
Could H-A include a detection of a Razberry device when installed on its GPIO pins and respond to the user to modify the config.txt or, even better, to run  a scrpit to amend the file and re-boot?  
It would need to include a warning that bluetooth is disabled unless it can be enabled using other pins.

pvizeli, 2020-04-24:
We should warn user to use raspberry. All other board do a better job in that case and user straggle all time over the same issues which are better solved by other boards.