home-assistant/operating-system #576: odroid N2 changes MAC-address after reboot

kreisklasse, 2020-02-09:
**Home Assistant release with the issue:**
0.105.2

**Operating environment (HassOS/Generic):**
HassOS 4.2 (same problem with version 4.0)
Supervisor 199
Deployment | development

**Supervisor logs:**
nil

**Network Hardware**
Unifi USG with Unifi Router

**Description of problem:**
Fresh install and rebooting odroid N2 via `Supervisor-System-Reboot`
odroidN2 shows up on LAN with a different MAC-address after each reboot.
So it gets a different IP each time. Tried it abou 6 times and got 6 new MAC-addresses.

pvizeli, 2020-02-09:
Fix with OS 4.2

kreisklasse, 2020-02-09:
@pvizeli 
i am running HassOS 4.2/Supervisor 199 and restarted the odroidN2 via `Supervisor-System-Reboot` after i read your post above.

But the odroidN2 got a new mac-address again. :-/

Here my System Log after rebooting:   https://paste.debian.net/1129864/

Any other logs i can post here to help ?

pvizeli, 2020-02-09:
What is the output of `ip addr show eth0` on host?

kreisklasse, 2020-02-09:
> ip addr show eth0

`~ $ ip addr show eth0
13: eth0@if14: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue state UP
link/ether 02:42:ac:1e:21:01 brd ff:ff:ff:ff:ff:ff
    inet 172.30.33.1/23 brd 172.30.33.255 scope global eth0
       valid_lft forever preferred_lft forever`

kreisklasse, 2020-02-09:
> 
> 
> What is the output of `ip addr show eth0` on host?

`~ $ ip addr show eth0
13: eth0@if14: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue state UP
    link/ether 02:42:ac:1e:21:01 brd ff:ff:ff:ff:ff:ff
    inet 172.30.33.1/23 brd 172.30.33.255 scope global eth0
       valid_lft forever preferred_lft forever`

pvizeli, 2020-02-09:
Do you see a  local mac address node on your device tree? They is on /proc/device-tree/soc/ethernet@..../

If not, look like on OTA update the bootloader was not updated. Can you reflash the device?

kreisklasse, 2020-02-09:
> Do you see a local mac address node on your device tree? They is on /proc/device-tree/soc/ethernet@..../

This i don't understand. Do you mean the mac-address of the N2 ? It is now 3a:35:37:d4:fd:32

> 
> If not, look like on OTA update the bootloader was not updated. Can you reflash the device?

Already reflashed the SD-card for the N2 with the "hassos_odroid-n2-4.0.img.gz"-Image.
I rebootet it with HassOS 4.0 running, and it always got a new mac-address.
Then updated to HassOS 4.2 via the `Supervisior-Dashboard` where it showed the new 4.2-version.
And it still got new mac's after reboot.

pvizeli, 2020-02-09:
What is the output of https://www.kernel.org/doc/html/latest/driver-api/nvmem.html#userspace-binary-interface with `cat`.

And I need the output of your device-tree node of your network card.

I think on the update from 4.0 to 4.2 it did not update the bootloader and without new bootloader you have the issue of changing mac.

pvizeli, 2020-02-09:
https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/misc/rauc-hook

Yes, they did not update the bootloader, I need fix that for next version. To fix the issue, you need reflash the device or wait until next version

kreisklasse, 2020-02-09:
> 
> 
> What is the output of https://www.kernel.org/doc/html/latest/driver-api/nvmem.html#userspace-binary-interface with `cat`.

hope it is this:
`~ $ cat /sys/bus/nvmem/devices/*/nvmem`
Output :
`a70fe3df067249e3ac26001e06423cd3`

> And I need the output of your device-tree node of your network card.

Sorry, i have no clue what the means and how to get it.

I will wait ;-)

PR #578 (2020-02-10): Change handling for SPL based images
Fix: #576
closed 2020-02-10 by pvizeli