home-assistant/operating-system #583: bootchooser not working (USB boot on NUC)

SamJongenelen, 2020-02-13:
**HassOS release with the issue:**
3.10 stable

when booting from usb (wrote the Etcher image); the Hassos boot menu is shown. However, booting is not working..
Am i doing something wrong?  I've updated bios, formatted the SSD to NTFS.

[image]

marine1988, 2020-03-31:
same problem here!

SamJongenelen, 2020-03-31:
FIY: I ended up with Ubuntu with Docker

pvizeli, 2020-05-08:
It just works over a native UEFI block storage protocol.

closed 2020-05-08 by pvizeli

ericlamotte, 2020-05-11:
This is happening to me on my NUC7i3BNH (updated BIOS) with HassOS 3.13. Sometimes I can get it to work for a few boots before it stops booting. Can you explain more, pvizeli?

BigDoom, 2020-12-19:
Had the exact same issue; this post helped me: https://community.home-assistant.io/t/my-steps-to-success-for-hass-io-on-intel-nuc/43915/356

What I ended up doing was to take out the SSD from my NUC, connect it to my laptop, directly load the HassOS image for NUC onto the SSD, but it back into my NUC, booted and everything started like it should. 

Perhaps it works for you as well.