home-assistant/operating-system #586: Enable HassOS i2c not working via import USB

Frederik-V, 2020-02-16:
Dear.

I want to make use of the MCP23017 I/O Expander integration to be able to start using the 32 channel expansion HAT (https://thepihut.com/products/modmypi-mcp23017-hat-32-channel-io-expansion-hat) with HASSOS in combination with RPI 3b.

I have followed several times the described steps (see https://www.home-assistant.io/hassio/enable_i2c) by using the USB import, but so far without any success :-(
There is also no feedback towards the user if the USB import is really performed as it should.

Is there a certain workaround or another smart way to enable the I2C bus on the RPI 3b with HASSOS?
Or a step further, is it not possible to introduce a sort of dashboard where you can enable/visualize RPI features (I2C, GPIO, audio, etc.) in one view with direct feedback to the user if the feature is enabled or not (green=on/red=off or gray=off)?

Many thanks for the support on this item.

thecreaser, 2020-03-09:
I also am having trouble getting my Pi3B+ reading the GPIO pins. I also find the USB import does not work for me.  I'm a noob, so I might be getting some of this wrong....

I found this information:
https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/raspberrypi.md#i2c

> **I2C**
> Add _dtparam=i2c1=on_ and _dtparam=i2c_arm=on_ to config.txt. After that we create a module file on host (with config usb stick or) direct into /etc/modules-load.d.
> 
> **rpi-i2c.conf:**
> 
> _i2c-dev_
> _i2c-bcm2708_

To clarify:
1) I created a new folder /etc/modules-load.d
2) I created a new file in modules-load.d called rpi-i2c.conf
3) I added the two lines of code to rpi-i2c.conf

It is not clear to me where the the config.tx file should go.  I am experimenting.

Kentrp, 2020-03-22:
put your sd card (with HASSOS) to usb on linux or windows (MacOs don't see card) systems, there youll see config.tx file

gralin, 2020-04-26:
Thank you @Kentrp for your tip, this did the trick for me. I agree with everything @Frederik-V wrote, I wasted a lot of time looking in wrong places, but I think it's the problem is USB config not being imported. I had authorized_keys to enable SSH on the same USB stick and that worked no problem, so it's just issue with config.txt apparently.

closed 2020-06-30 by cogneato