home-assistant/operating-system #590: Reopen issue #224

mattclar, 2020-02-18:
@pvizeli  Please reopen issue #224 
There is active issues from multiple users but it was closed by the stale-bot :/

rpitera, 2020-02-18:
In the meantime, I'll add my voice to the chorus from 224 and say I've had this same issue for at least 10 releases.

cogneato, 2020-03-03:
That issue covers half a dozen different unrelated issues and a lot of misunderstanding. Some specifics would be helpful.

blamaz, 2020-03-03:
I've restarted from scratch and, while reinstalling my add-ons, this error appeared in the system log right after adding the Mosquitto MQTT broker. I'm running hassos_ova-3.11 in virtual box.
ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused]. This error is appearing since a long time but my installation seem to be working ok

mnaz, 2020-03-26:
Not exactly the same it seems, but I get this in the system logs. My system is running fine and I just have been actively reviewing logs to check for any potential issues. I also have not enabled or setup ssl on anything at this point.

**ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].**

Proxmox 5.4-13

arch | x86_64
dev | false
docker | true
hassio | true
os_name | Linux
os_version | 4.19.107
python_version | 3.7.7
version | 0.107.7
virtualenv | false

Add-ons:
ESPHome 1.14.3
Mosquitto broker 5.1
TasmoAdmin 0.10.0
Samba share 9.0 (Not Active)
Hass.io Google Drive Backup 0.100.0 
Visual Studio Code 2.1.0

mnaz, 2020-04-06:
I have not seen this in the logs since I have updated to HassOS 3.12

chali272, 2020-04-10:
getting this 20-04-10 13:18:36 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].

dibyadip-das, 2020-05-07:
I am also getting the same error. Below are my logs. I am running the later HASSIO install of HomeAssistant 0.109.4 in my Raspberry PI 4

`20-05-07 08:51:27 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:27 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:27 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:32 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:32 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:32 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:37 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:37 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:37 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:42 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:42 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:42 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:47 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:47 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:47 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:52 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:52 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:52 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 08:51:56 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
20-05-07 08:51:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 08:51:57 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 08:51:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-05-07 08:57:31 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-07 08:58:44 INFO (MainThread) [supervisor.homeassistant] Updated Home Assistant API token
20-05-07 09:11:38 INFO (MainThread) [supervisor.api.security] /homeassistant/info access from cebe7a76_hassio_google_drive_backup
20-05-07 09:11:38 INFO (MainThread) [supervisor.api.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
20-05-07 09:12:02 INFO (MainThread) [supervisor.api.security] /snapshots access from cebe7a76_hassio_google_drive_backup
20-05-07 09:16:39 INFO (SyncWorker_1) [supervisor.docker.interface] Restart homeassistant/raspberrypi4-homeassistant
20-05-07 09:16:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
20-05-07 09:16:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-05-07 09:16:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:16:46 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 09:16:46 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 09:16:51 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:16:51 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 09:16:51 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 09:16:56 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:16:56 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 09:16:56 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 09:17:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:17:01 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 09:17:01 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 09:17:06 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:17:06 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
[... cut ...]
20-05-07 09:18:32 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-05-07 09:18:37 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-05-07 09:18:37 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-05-07 09:18:38 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-05-07 09:18:40 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance`

brandon-page, 2020-05-17:
Same here running natively on Ubuntu 20.04.  Home Assistant 0.109.6.  Supervisor 222.

thehaxxa, 2020-05-20:
Same here on Ubuntu 20.20

mkanet, 2020-05-23:
I have a Ubuntu (VMWare) VM for HASSOS.  I'm using the latest version of Home Assistant (version .110).  

My VM sometimes fails to boot.  Also I get the error below in my log too (seems to be related to the MQTTT broker addon); although, I have a MQTT client successfully communicating with the broker.

`
20-05-23 03:45:56 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused]
`

So.... what's the trick to get this to working correctly?   Are there any developers even looking into this issue?  Seems like a long time to have this issue without a fix (or at least a work-around).

mattbruman, 2020-07-10:
Having the same issue for awhile now, I cant pinpoint it excet it happened after updating to .111

`20-07-10 11:59:58 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:08:37 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:10:13 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:12:33 INFO (MainThread) [supervisor.homeassistant] Updated Home Assistant API token
20-07-10 12:16:00 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:42:38 INFO (MainThread) [supervisor.homeassistant] Updated Home Assistant API token
20-07-10 12:53:32 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:54:59 INFO (MainThread) [supervisor.api.security] /snapshots access from cebe7a76_hassio_google_drive_backup
20-07-10 12:55:28 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 12:59:57 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:07:03 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:08:39 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:12:26 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:14:02 INFO (MainThread) [supervisor.homeassistant] Updated Home Assistant API token
20-07-10 13:18:42 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:21:59 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-07-10 13:23:35 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: `

nathanle, 2020-07-30:
Also having the issue. I wish I had data to give, but I did not enable SSH. However, the pi is up, I can ping it, but nnap and manual TCP connections to 8123 are fruitless (see what I did there). 

$ nmap 192.168.0.3 -p 8123
Starting Nmap 7.70 ( https://nmap.org ) at 2020-07-30 18:32 EDT
Nmap scan report for 192.168.0.3
Host is up (0.00033s latency).

PORT     STATE  SERVICE
8123/tcp closed polipo

$ telnet 192.168.0.3 8123
Trying 192.168.0.3...
telnet: Unable to connect to remote host: Connection refused

Also, I only have Z-Wave and Tile add-ons installed.

wolfjuli, 2020-08-17:
Same here,
using HassIO on Raspberry Pi 3 b

my .storage/auth looks like this:

`
{
                "access_token_expiration": 1800.0,
                "client_icon": null,
                "client_id": null,
                "client_name": null,
                "created_at": "2020-08-01T14:42:25.553564+00:00",
                "id": "dfe15aXXXX",
                "jwt_key": "c49a2XXXXdeb",
                "last_used_at": "2020-08-04T17:08:25.342789+00:00",
                "last_used_ip": "172.30.32.2",
                "token": "11edb3XXXX508",
                "token_type": "system",
                "user_id": "c74XXXXe4a8"
            },
            {
                "access_token_expiration": 1800.0,
                "client_icon": null,
                "client_id": "http://homeassistant.local:8123/",
                "client_name": null,
                "created_at": "2020-08-01T15:11:50.062752+00:00",
                "id": "49c1XXXXX8c",
                "jwt_key": "dc4121XXXX2429",
                "last_used_at": "2020-08-01T16:21:46.019764+00:00",
                "last_used_ip": "192.168.20.15",
                "token": "21a9XXXXe155",
                "token_type": "normal",
                "user_id": "ca16XXXXX798a"
            },
`

It seems like, something is registering as logged in, although it is faulty. I can't restart hassio because of that, and I'm getting the error "Failed to load panel source entrypoint.js" which I can't fix because of that. I'm not aware of any power outage or any other external influences and if I remove the entry from the auth file manually, it just immediately reappears.

guciopl, 2020-08-28:
I think it comes from mqtt addon.

20-08-28 01:38:14 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-08-28 01:38:14 INFO (MainThread) [supervisor.auth] Home Assistant not running, check cache
20-08-28 01:38:14 INFO (MainThread) [supervisor.auth] Cache hit for mqtt
20-08-28 01:38:24 INFO (MainThread) [supervisor.auth] Auth request from core_mosquitto for mqtt
20-08-28 01:38:24 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config: 
20-08-28 01:38:24 INFO (MainThread) [supervisor.auth] Home Assistant not running, check cache
20-08-28 01:38:24 INFO (MainThread) [supervisor.auth] Cache hit for mqtt
20-08-28 01:38:26 INFO (MainThread) [supervisor.auth] Success login from mqtt

Home Assistant 0.114.4 running on RPI4 with 3 addons:
Mosquitto broker
Samba share
zigbee2mqtt

mkanet, 2020-08-28:
@guciopl, I don't have any issues using the MQTT broker addon when using it with EventGhost MQTT.  It seems all the issues reported are revolved around IOT Link MQTT communications.

ygreq, 2020-09-28:
I just installed IOTLink a few days ago and I see now that I have the same problem. I also did some other things so could be from something else as well.

perkins1724, 2020-10-04:
+1 that it seems somehow tied to MQTT. I had same symptoms (Client error on WebSocket API) - reconfiguring the MQTT Integration cleared the error and allowed the update to proceed.

agners, 2020-11-03:
Is this still reproducible for someone? It seems this is more a MQTT Add-On related issue?

There are also two issues brought up here:

```
ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused
```

and

```
20-07-10 13:07:03 ERROR (MainThread) [supervisor.homeassistant] Error on call http://172.30.32.1:8123/api/config:
```

I am inclined closing this as this is unlikely a Operating System issue.

ludeeus, 2020-11-03:
You can get both those if the core container is not properly working (like bad config), it should not be in this repo @agners 
If anything it should be in the core repo, but Discord/forum would be better.