home-assistant/operating-system #592: Updating operating system Always stop my RPI 3B+

chris669, 2020-02-19:
Running hassos 3.10 on RPI 3B+.
Booting from USB, no SD.
Running HA Latest version 0.105.5
Each time I try to update, or click on update Button, system Shutdown. I have to unplug my RPI in order to reboot. 
This is not new, have this since I boot from USB. I've seen there is an issue with SSD, but as mine is with USB...

pvizeli, 2020-02-19:
I the same issue. Not sure if that is an UI bug. Can you use the CLI for update?

chris669, 2020-02-19:
Hi, never tried the CLI, except for reloading HA. Could you please give me the command syntax? (sorry do not know it...) and will try next time.

vdwkes, 2020-02-19:
got the same problem yesterday, today the update worked

closed 2020-04-26 by stale[bot]