home-assistant/operating-system #593: Can't fetch versions

allmoney, 2020-02-19:
I installed HA to pi4 and long time can’t update from Operating System 3.3 to any new version. Now to 3.10.
Got error:
```
20-02-15 15:55:28 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/3.10/hassos_rpi4-3.10.raucb: 
20-02-15 15:55:28 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<HassOS.update() done, defined at /usr/src/hassio/hassio/hassos.py:149> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/hassio/hassio/hassos.py", line 160, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/hassio/hassio/hassos.py", line 110, in _download_raucb
    raise HassOSUpdateError()
hassio.exceptions.HassOSUpdateError
```
How fix it? How update? Please, help!

[image]

allmoney, 2020-02-19:
Was fixed in 3.11
Thanks!

closed 2020-02-19 by allmoney

KM192, 2020-02-19:
I'am trying to upgrade from 3.7 to 3.11 and still having error:
20-02-19 18:06:24 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/3.11/hassos_rpi4-3.11.raucb: 
20-02-19 18:06:24 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<HassOS.update() done, defined at /usr/src/hassio/hassio/hassos.py:150> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/hassio/hassio/hassos.py", line 161, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/hassio/hassio/hassos.py", line 111, in _download_raucb
    raise HassOSUpdateError()
hassio.exceptions.HassOSUpdateError