home-assistant/operating-system #595: FTP Add-on pulling incorrect time information for file modification info

JimmyJones79, 2020-02-20:
I already posted this on the FTP add-on Git and was directed here from the creator since his add-on gets it's time info from Hass.

I am on the East coast of the US in the Easter Time Zone, my Raspberry Pi 3b+ is running the latest updated version of Hass.io (Home Assistant).

For some reason the FTP Add-on is showing file creation times as if it was Hawaiian time instead of East Coast time. This is weird because Hassio and the Raspberry OS are showing the correct time.

**Expected behavior**
Show the file creation time as Eastern Time

**Actual behavior**
file creation time is the Hawaiian time zone

See the screen shot attached.  On the left you will see Hass's time at 15:01 (3:01pm) on the right you will see the last directory was created at 10:00am, However this just happened at 3pm.  For some reason the FTP server is using Hawaiian time.

[image]

pvizeli, 2020-02-20:
This is for Home Assistant OS and not Raspberry OS. Please open a issue on Raspberry Pi

closed 2020-02-20 by pvizeli

JimmyJones79, 2020-02-21:
It is Home Assistant OS running on a Raspberry Pi 3b+